rerun-io/rerun #7298: Native Video support

emilk, 2024-08-28:
We'll soon have experimental video support in the web viewer (https://github.com/rerun-io/rerun/issues/7271).

The next step is to get that working on native too.

jprochazk, 2024-10-02:
(Note that I made the same mistake as in `re_mp4` and uploaded a bunch of binary files to `rav1d-rs`, which need cleaning up via history rewriting and either be turned into LFS files or removed)

emilk, 2024-10-03:
@jprochazk I summarized my understanding of `rav1d-rs` in https://github.com/rerun-io/rerun/issues/7575 - feel free to edit/comment

Wumpf, 2025-01-27:
There's always more codecs we'd like to support, but the ones listed here are all in!

For a full overview what we support check
https://rerun.io/docs/reference/video#codec-support

closed 2025-01-27 by Wumpf