rerun-io/rerun #7337: Blueprint API: Visibility per entity basis(python)

NickLaurenson-Visionick, 2024-09-03:
**Is your feature request related to a problem? Please describe.**

Currently it's not possible to set the visibility per entity in the python blueprint API. 
The current workaround is to use the per view `content` keyword to remove the data from the view.

**Describe the solution you'd like**

Maybe add a visibility keyword in the View API

**Describe alternatives you've considered**

**Additional context**

Mimic this functionnality from the UI:
[image]

Wumpf, 2024-09-09:
Associated discord thread https://discord.com/channels/1062300748202921994/1280068537666240584/1280068537666240584

Overrides of (individual) visibility is already possible, but slightly awkward and not documented:

```py
import rerun as rr
import rerun.blueprint as rrb

rr.init(
    "rerun_example_visibility_via_blueprint",
    spawn=True,
    default_blueprint=rrb.Spatial3DView(
        overrides={"points": [rrb.components.Visible(False)]}
    ),
)
rr.log("points", rr.Points3D([[0, 0, 0], [1, 1, 1]]))
```

Should document this and make it a bit more approachable maybe

kabouzeid, 2024-11-21:
Ideally, we could also set the default visibility during logging. Currently I'm logging many entity paths, and have to manually click 20+ times to hide the stuff I might want to look at on demand, but usually needs to be hidden.