rerun-io/rerun #7346: Enhance Support for Custom Axes and 2D Plotting

pperanich, 2024-09-03:
**Is your feature request related to a problem? Please describe.**

I'm currently using the rerun viewer to visualize time series data. However, I'm facing challenges when trying to visualize the frequency spectrum of a time series signal. The `SeriesLine` plot is limited by requiring the x-axis to be a time axis, which isn't suitable for frequency spectrum data where the x-axis should represent frequency. Additionally, the `LineStrip2D` plot lacks axes and grids, and the `BarChart` lacks support for custom x-values, making these options less ideal for frequency spectrum visualization.

**Describe the solution you'd like**

I propose the introduction of a new plot type or an enhancement to the existing `SeriesLine` plot that allows for frequency spectrum visualization. Key features should include:
- Support for custom x-values, enabling the x-axis to represent frequency bins instead of time.
- Axes and grid support.
- The ability to plot the data as a continuous line strip or as points (e.g., using `SeriesPoint`), with customizable markers.

**Describe alternatives you've considered**

- Using the `BarChart` with custom x-values (once available) to represent the frequency spectrum.
- Using `LineStrip2D` without any axes information.

**Additional context**

The ability to visualize frequency spectra directly within rerun would significantly enhance its utility for multi-modal time series analysis.

Wumpf, 2024-09-09:
Related to:
* https://github.com/rerun-io/rerun/issues/5622
* https://github.com/rerun-io/rerun/issues/1387

I was sure there was already a ticket on general 2D plotting itself, but couldn't find it, so might as well be the first! :)

abey79, 2024-12-03:
I believe this proposal would fit this particular need:
- https://github.com/rerun-io/rerun/issues/8286

Xiao-Chenguang, 2025-07-07:
Same issue around custom x-axis in BarChart. Internally, rerun use `egui_plot` which support custom x-axis to draw BarChart. I tried to enable feature as @pperanich suggested:
>     * Using the `BarChart` with custom x-values (once available) to represent the frequency spectrum.

Check the [fork](https://github.com/Xiao-Chenguang/rerun) for more detail.

Example code in rust:
```rust
fn main() -> Result<(), Box<dyn std::error::Error>> {
    let rec = rerun::RecordingStreamBuilder::new("custom_index_barchart").spawn()?;
    let x = vec![1, 2, 4, 8];
    let y = vec![4, 5, 3, 6];
    rec.log(
        "bars",
        &rerun::BarChart::new(x, y).with_color(rerun::Color::from_rgb(200, 80, 123)),
    )?;

    Ok(())
}
```

The results looks like:
[image]