rerun-io/rerun #7359: Reimplement `pixi run py-wheel`

abey79, 2024-09-05:
As per our current `BUILD.md`, users should be able to locally build wheels using `pixi run py-wheel --release`. This command is for some reason no longer in our `pixi.toml` file. This should be fixed.

- Would have been useful for (at least) [this user](https://discord.com/channels/1062300748202921994/1268203898296991766/1268203898296991766).
- Helps with https://github.com/rerun-io/rerun/issues/2628 (by making it easier to build wheel compatible with the local architecture).

Wumpf, 2024-09-09:
See also
* https://github.com/rerun-io/rerun/pull/6490

jleibs, 2024-10-28:
The most annoying part of this is the instructions needs some careful per-platform handling.  Here's the sequence of commands to run on linux in the meantime:
```
pixi run rerun-build-native-and-web-release
cp target/release/rerun rerun_py/rerun_sdk/rerun_cli/
pixi run python scripts/ci/build_and_upload_wheels.py  --dir unused --target x86_64-unknown-linux-gnu --compat manylinux_2_34
```