rerun-io/rerun #7360: Option to Disable Background Grid in TimeSeriesView

michel-ristau-st, 2024-09-05:
**Description:** 

It would be helpful to have an option to disable the background grid in the `TimeSeriesView`. While the grid can be useful for certain visualizations, there are cases where a cleaner background without the grid would be more appropriate, especially in presentations or reports where the focus is solely on the data.

**Proposed Feature:**

- Add a toggle option (e.g., `show_grid: bool`) in the `TimeSeriesView` settings.
- Default could be set to `True` (grid enabled) for backwards compatibility, but allowing users to disable it by setting it to `False`.

**Use Case:**

This feature would be useful for users who prefer a minimalist view without the distraction of grid lines, especially for cleaner visualization in certain contexts, such as during data analysis presentations or embedded plots in reports.

**Benefits:**

- Greater flexibility in customizing the appearance of time series charts.
- Improved readability for users who prefer a less cluttered visualization.

grtlr, 2025-07-16:
In a similar vain, the background should be adjustable as well. I'm changing the title of the issue to reflect that.

michel-ristau-st, 2025-09-22:
It looks like you've implemented all the desired features for 3D views (rerun version 0.24.0).

[image]

It would be great if these settings could also be configured in the plots.