rerun-io/rerun #7372: Add helpful error message on WebGPU error

emilk, 2024-09-09:
Some users report seeing this in their browser:

[image]

The solution to this problem is to use `https://rerun.io/viewer?renderer=webgl` as the URL. That should be part of the error message. So we should:

A) Improve error message
B) Make sure the error appears in the browser (not just in the dev console)

Or better yet: automatically switch to the WebGL backend if WebGPU fails

Wumpf, 2024-09-09:
This should actually already display the default screen for rendering related failures, including an explainer and link to switch the backend to webgl.
Do we have any reason to believe that's not the case here?

Note that the particular error is caused by an outdated chrome version.

emilk, 2024-09-10:
Replaced with:
* https://github.com/rerun-io/rerun/issues/7392

closed 2024-09-10 by emilk