rerun-io/rerun #7373: Warn on unsupported video

jprochazk, 2024-09-09:
When a user loads a video which we can't load for some reason (failed to load, unsupported container format, unsupported codec, etc.), we should report that to them in some way other than a short-lived warning notification.

gavrelina, 2024-09-09:
A ui/ux proposal! 
- highlight the component in the stream that causes an error to improve its discoverability
- in the error copy itself mention explicitly what's wrong and why and how users can recover from it (i.e. linking to github issue to follow or docs)
- visually display smth explicitly "broken" instead of the video in the Viewer, like this:

[image]

gavrelina, 2024-09-10:
@Wumpf makes a lot of sense! 

About the entity msg => it would also then be visible if users click on it via the blueprint, right? that would also largly increase the discoverability! I do worry (slightly) that copying it in few places comes across as "omg! so many errors everywhere, panic!", hence my initial idea was to only show at Blob level and highlight Blob in red so that users would go there to find it. so kinda direct them with color nudging :) what do you think? 

And moving forward from here it still makes sense to revise (revive?) the different gradations and make this thing work : https://github.com/rerun-io/rerun/issues/4304

jprochazk, 2024-09-11:
From: https://github.com/rerun-io/rerun/pull/7398#discussion_r1753388072

> I don't know if you've experienced this scenario: You're playing a video on the web, and it suddenly fails with an error of some kind. The player just stops, and the controls become inactive, so you can't seek to a different point in the video anymore. But then you reload the page, and seek after the point where the player previously stopped with an error, and the video continues playing back just fine.
> 
> I think we could do a lot better than web video players here, by always allowing seek even in the case of an error, and more importantly also highlight which segments are broken on the timeline, so the user knows exactly where to seek next to try and catch a part of the video which _can_ be played back.
> 
> This kind of behavior would definitely follow our "handle anything the user throws at it" philosophy

gavrelina, 2024-09-11:
@jprochazk I can already picture how we mark some specific events/part on the timeline where one encounters an "error"!

commit 30f28776 (2024-09-19): Display video errors & video blob information (#7433)

### What

* Fixes #7373
* I'm sure there's more to do here and surely more cases to try out
whether they display the correct errors, but this PR should handle
everything that we detect as an error case, laying the foundation to
pass everything through

As part of this work it was necessary to make clearer difference between
video loading and video decoding. You'll notice changes in re_renderer's
video and the video cache accordingly.
(there's some loose ends here like "don't create a decoder when we're
just inspecting a blob" which will be [... cut]
closed 2024-09-19 by Wumpf