rerun-io/rerun #7384: VS 2017 Compiler support for Rerun SDK

benliepert, 2024-09-10:
I'm attempting to integrate Rerun into a C++ project that uses the VS 2017 compiler (version 19.16.27031.1 for x86). I followed the standard cmake setup as documented, but got a build error in src/rerun/error.cpp:
`error C2039: 'tolower': is not a member of 'std'`
This is the relevant LOC:
`        std::transform(v.begin(), v.end(), v.begin(), [](char c) { return std::tolower(c); });`

This is resolved by adding an include for cctype in this file.
`#include <cctype>`

This error is reproducible using a blank project & standalone file. The header inclusion behavior must have changed by VS2022.

This is all that I needed to change for the project to compile (and I validated a minimal example in the viewer)

I realize VS 2017 is fairly old (there's VS2019), and this is easy to work around. But I want to document it in case it aids the decision of whether to test against/support VS 2017 going forward.

Wumpf, 2024-09-10:
Is this the only issue you're running into when using vs2017?
We only test VS 2022 on CI so far, but [apparently](https://learn.microsoft.com/en-us/cpp/overview/visual-cpp-language-conformance) VS2017 (MSVC 19.15) has full C++17 support so we should be able to make it run. I'd be happy to add official support but only if we can add it to the CI test matrix and it doesn't turn out to be a maintenance nightmare with little usage :). Contributions very much welcome!

(what we're testing on is documented here https://ref.rerun.io/docs/cpp/stable/#autotoc_md16)

benliepert, 2024-09-12:
Yes, this is the only issue I've had. Only played around with logging a few hundred scalars at 1hz