rerun-io/rerun #7385: ERROR eframe::native::run] Exiting because of error: app creation error

DoongLi, 2024-09-10:
Hi, my environment is Ubuntu 20.04, x86, computer CPU integrated graphics, no independent graphics card.
I use ‘pip3 install rerun-sdk’ install rerun，but when I use ‘rerun’ to start it, some errors happened below:
[2024-09-10T01:26:43Z INFO  re_sdk_comms::server] Hosting a SDK server over TCP at 0.0.0.0:9876. Connect with the Rerun logging SDK.
[2024-09-10T01:26:43Z INFO  winit::platform_impl::platform::x11::window] Guessed window scale factor: 1
[2024-09-10T01:26:43Z WARN  wgpu_hal::vulkan::instance] Unable to find extension: VK_EXT_swapchain_colorspace
libEGL warning: DRI2: failed to authenticate
[2024-09-10T01:26:43Z WARN  wgpu_hal::gles::adapter] Returned GL context is 3.1, when 3.3+ is needed
[2024-09-10T01:26:43Z WARN  wgpu_hal::gles::adapter] Returned GL context is 3.1, when 3.3+ is needed
[2024-09-10T01:26:43Z WARN  wgpu_core::instance] Missing downlevel flags: DownlevelFlags(SURFACE_VIEW_FORMATS)
    The underlying API or device in use does not support enough features to be a fully compliant implementation of WebGPU. A subset of the features can still be used. If you are running this program on native and not in a browser and wish to limit the features you use to the supported subset, call Adapter::downlevel_properties or Device::downlevel_properties to get a listing of the features the current platform supports.
[2024-09-10T01:26:43Z WARN  wgpu_core::instance] DownlevelCapabilities {
        flags: DownlevelFlags(
            COMPUTE_SHADERS | FRAGMENT_WRITABLE_STORAGE | INDIRECT_EXECUTION | BASE_VERTEX | READ_ONLY_DEPTH_STENCIL | NON_POWER_OF_TWO_MIPMAPPED_TEXTURES | CUBE_ARRAY_TEXTURES | COMPARISON_SAMPLERS | INDEPENDENT_BLEND | VERTEX_STORAGE | FRAGMENT_STORAGE | MULTISAMPLED_SHADING | DEPTH_TEXTURE_AND_BUFFER_COPIES | WEBGPU_TEXTURE_FORMAT_SUPPORT | BUFFER_BINDINGS_NOT_16_BYTE_ALIGNED | UNRESTRICTED_INDEX_BUFFER | FULL_DRAW_INDEX_UINT32 | DEPTH_BIAS_CLAMP | VIEW_FORMATS | UNRESTRICTED_EXTERNAL_TEXTURE_COPIES | NONBLOCKING_QUERY_RESOLVE | VERTEX_AND_INSTANCE_INDEX_RESPECTS_RESPECTIVE_FIRST_VALUE_IN_INDIRECT_DRAW,
        ),
        limits: DownlevelLimits,
        shader_model: Sm5,
    }
WARNING: lavapipe is not a conformant vulkan implementation, testing use only.
[2024-09-10T01:26:43Z WARN  re_renderer::context] Bad software rasterizer detected - expect poor performance and crashes. See: https://www.rerun.io/docs/getting-started/troubleshooting#graphics-issues
[2024-09-10T01:26:43Z INFO  re_renderer::context] wgpu adapter backend: Vulkan, device_type: Cpu, name: "llvmpipe (LLVM 12.0.0, 256 bits)", driver: "llvmpipe", driver_info: "Mesa 21.2.6 (LLVM 12.0.0)"
[2024-09-10T01:26:43Z ERROR eframe::native::run] Exiting because of error: app creation error: The given device doesn't support the required downlevel capabilities for the given hardware caps DeviceCaps { tier: FullWebGpuSupport, max_texture_dimension2d: 16384, max_buffer_size: 2147483647, backend_type: WgpuCore }.
    Required: DownlevelCapabilities { flags: DownlevelFlags(COMPUTE_SHADERS | FRAGMENT_WRITABLE_STORAGE | INDIRECT_EXECUTION | BASE_VERTEX | READ_ONLY_DEPTH_STENCIL | NON_POWER_OF_TWO_MIPMAPPED_TEXTURES | CUBE_ARRAY_TEXTURES | COMPARISON_SAMPLERS | INDEPENDENT_BLEND | VERTEX_STORAGE | ANISOTROPIC_FILTERING | FRAGMENT_STORAGE | MULTISAMPLED_SHADING | DEPTH_TEXTURE_AND_BUFFER_COPIES | WEBGPU_TEXTURE_FORMAT_SUPPORT | BUFFER_BINDINGS_NOT_16_BYTE_ALIGNED | UNRESTRICTED_INDEX_BUFFER | FULL_DRAW_INDEX_UINT32 | DEPTH_BIAS_CLAMP | VIEW_FORMATS | UNRESTRICTED_EXTERNAL_TEXTURE_COPIES | SURFACE_VIEW_FORMATS | NONBLOCKING_QUERY_RESOLVE | VERTEX_AND_INSTANCE_INDEX_RESPECTS_RESPECTIVE_FIRST_VALUE_IN_INDIRECT_DRAW), limits: DownlevelLimits, shader_model: Sm4 }
    Actual: DownlevelCapabilities { flags: DownlevelFlags(COMPUTE_SHADERS | FRAGMENT_WRITABLE_STORAGE | INDIRECT_EXECUTION | BASE_VERTEX | READ_ONLY_DEPTH_STENCIL | NON_POWER_OF_TWO_MIPMAPPED_TEXTURES | CUBE_ARRAY_TEXTURES | COMPARISON_SAMPLERS | INDEPENDENT_BLEND | VERTEX_STORAGE | FRAGMENT_STORAGE | MULTISAMPLED_SHADING | DEPTH_TEXTURE_AND_BUFFER_COPIES | WEBGPU_TEXTURE_FORMAT_SUPPORT | BUFFER_BINDINGS_NOT_16_BYTE_ALIGNED | UNRESTRICTED_INDEX_BUFFER | FULL_DRAW_INDEX_UINT32 | DEPTH_BIAS_CLAMP | VIEW_FORMATS | UNRESTRICTED_EXTERNAL_TEXTURE_COPIES | NONBLOCKING_QUERY_RESOLVE | VERTEX_AND_INSTANCE_INDEX_RESPECTS_RESPECTIVE_FIRST_VALUE_IN_INDIRECT_DRAW), limits: DownlevelLimits, shader_model: Sm5 } during event Resumed
Error: app creation error: The given device doesn't support the required downlevel capabilities for the given hardware caps DeviceCaps { tier: FullWebGpuSupport, max_texture_dimension2d: 16384, max_buffer_size: 2147483647, backend_type: WgpuCore }.
Required: DownlevelCapabilities { flags: DownlevelFlags(COMPUTE_SHADERS | FRAGMENT_WRITABLE_STORAGE | INDIRECT_EXECUTION | BASE_VERTEX | READ_ONLY_DEPTH_STENCIL | NON_POWER_OF_TWO_MIPMAPPED_TEXTURES | CUBE_ARRAY_TEXTURES | COMPARISON_SAMPLERS | INDEPENDENT_BLEND | VERTEX_STORAGE | ANISOTROPIC_FILTERING | FRAGMENT_STORAGE | MULTISAMPLED_SHADING | DEPTH_TEXTURE_AND_BUFFER_COPIES | WEBGPU_TEXTURE_FORMAT_SUPPORT | BUFFER_BINDINGS_NOT_16_BYTE_ALIGNED | UNRESTRICTED_INDEX_BUFFER | FULL_DRAW_INDEX_UINT32 | DEPTH_BIAS_CLAMP | VIEW_FORMATS | UNRESTRICTED_EXTERNAL_TEXTURE_COPIES | SURFACE_VIEW_FORMATS | NONBLOCKING_QUERY_RESOLVE | VERTEX_AND_INSTANCE_INDEX_RESPECTS_RESPECTIVE_FIRST_VALUE_IN_INDIRECT_DRAW), limits: DownlevelLimits, shader_model: Sm4 }
Actual: DownlevelCapabilities { flags: DownlevelFlags(COMPUTE_SHADERS | FRAGMENT_WRITABLE_STORAGE | INDIRECT_EXECUTION | BASE_VERTEX | READ_ONLY_DEPTH_STENCIL | NON_POWER_OF_TWO_MIPMAPPED_TEXTURES | CUBE_ARRAY_TEXTURES | COMPARISON_SAMPLERS | INDEPENDENT_BLEND | VERTEX_STORAGE | FRAGMENT_STORAGE | MULTISAMPLED_SHADING | DEPTH_TEXTURE_AND_BUFFER_COPIES | WEBGPU_TEXTURE_FORMAT_SUPPORT | BUFFER_BINDINGS_NOT_16_BYTE_ALIGNED | UNRESTRICTED_INDEX_BUFFER | FULL_DRAW_INDEX_UINT32 | DEPTH_BIAS_CLAMP | VIEW_FORMATS | UNRESTRICTED_EXTERNAL_TEXTURE_COPIES | NONBLOCKING_QUERY_RESOLVE | VERTEX_AND_INSTANCE_INDEX_RESPECTS_RESPECTIVE_FIRST_VALUE_IN_INDIRECT_DRAW), limits: DownlevelLimits, shader_model: Sm5 }

Wumpf, 2024-09-10:
This sounds like you might have outdated drivers. What igpu (or cpu) are you running on?

> Returned GL context is 3.1, when 3.3+ is needed

This implies that your driver doesn't support OpenGL 3.3 which is the minimum required for the OpenGL fallback (Vulkan is preferred whenever possible) used by Rerun. 
It keeps trying but eventually doesn't find required features so switches to a Vulkan software rasterizer:
> wgpu adapter backend: Vulkan, device_type: Cpu, name: "llvmpipe (LLVM 12.0.0, 256 bits)", driver: "llvmpipe", driver_info: "Mesa 21.2.6 (LLVM 12.0.0)"

Only to then notice that this also doesn't have the required features.
`Mesa 21.2.6` is from 2021, so I'd start there trying to update that.

---

We need to find ways to make these errors more readable 🤔. It's tricky because there's so many ways of how these things can go wrong and it's pretty hard to reproduce.

DoongLi, 2024-09-10:
This is my configuration, and there is no independent graphics card.
[image]

DoongLi, 2024-09-10:
rerun's tutorial did not tell me that I need other dependencies. I don't know how to deal with this problem

Wumpf, 2024-09-10:
Ubuntu 20.04 is fairly old at this point and your hardware relatively new, so it's not too surprising that things don't work out of the box.
I'm no expert on this and don't have a repro case, but maybe try this guide to updating the mesa drivers
https://linuxcapable.com/how-to-upgrade-mesa-drivers-on-ubuntu-linux/

DoongLi, 2024-09-10:
I'll try it, I'll describe the whole process and if I solve it, thx, have a great day!

Wumpf, 2024-09-10:
well, until it's not confirmed fixed on your end and not better documented on ours this isuses is not solved

DoongLi, 2024-09-11:
I think it may be the reason of the graphics card, I changed a computer, the same is ubuntu 20.04, but this computer has 3060, no error.

Wumpf, 2024-09-11:
it should still run fine on that igpu. It has all the capabilities needed, as I wrote it's almost certainly a driver issue