rerun-io/rerun #7404: static=True image logging still stores all the data

nisseknudsen, 2024-09-11:
**Describe the bug**
I am using `rr.log(..., static=True)` with images and the expectation that the memory consumption should stay stable over time. I only want to see the latest logged image.
I also have the expectation that when opening the (web) viewer, I do not have to watch through all of the history to the current frame.
Both is not met for some reason.

**To Reproduce**
Script to repro (uses webcam `0` on macOS, but should work with any computer with we webcam. Maybe try `/dev/video0` for ubuntu).
[scratch_5.py.zip](https://github.com/user-attachments/files/16968845/scratch_5.py.zip)

**Expected behavior**
Sidebar data source size stays stable and whatever a single JPEG image takes, let's say 20mbyte to be generous.
Unfortunately, it doesn't

[image]

**Desktop (please complete the following information):**
 - OS: macOS Sonoma 14.6.1

**Rerun version**
Tested on 18.0 and 18.2 with same unexpected outcomes.

**Additional context**
Discord post: https://discord.com/channels/1062300748202921994/1075873257124810852/1283124563902992537

teh-cmc, 2024-09-23:
There are two issues at play:
1. Static chunks are not subject to the dangling detection/cleanup logic of the garbage collector, since static data as a whole is immune to garbage collection.
2. The decoded image cache doesn't subscribe to store events, and therefore continues caching stuff that has long been overwritten.

The fix for 1) is to properly account for static overwrites on the write path: patch all dangling references, decreases stats, fire events.
That should be a small amount of (likely tricky) code.

The fix for 2) is to set up a store subscriber so that the image cache drops overwritten data immediately.
That can range from trivial to extremely painful depending on how things are setup at the moment; I need to have a closer look.

closed 2024-09-27 by teh-cmc

nisseknudsen, 2024-10-18:
Thank you for the fix! Can confirm it works now 🙏

PR #7525 (2024-09-27): Invalidate hub-wide caches on deletions and overwrites
Hub-wide caches now subscribe to store events and invalidate accordingly in the face of deletions and overwrites.

This is a crutch to compensate for the lack of secondary caching, but a much needed crutch: the Rerun Viewer can now effectively be used as a soft realtime telemetry system.

https://github.com/user-attachments/assets/f1136f9d-e1fd-4e6b-87c6-422d5f3345e8

* Fixes #7404 

---

## Checklist

### `EncodedImage`

```python
for _ in range(0, 100):
    rr.log("image", rr.EncodedImage(path=image_file_path), static=True)
    time.slee [... cut]
commit f5aa0a09 (2024-09-27): Invalidate hub-wide caches on deletions and overwrites (#7525)

Hub-wide caches now subscribe to store events and invalidate accordingly
in the face of deletions and overwrites.

This is a crutch to compensate for the lack of secondary caching, but a
much needed crutch: the Rerun Viewer can now effectively be used as a
soft realtime telemetry system.

https://github.com/user-attachments/assets/f1136f9d-e1fd-4e6b-87c6-422d5f3345e8

* Fixes #7404 

---

## Checklist

### `EncodedImage`

```python
for _ in range(0, 100):
    rr.log("image", rr.EncodedImage(path=image_file_path), static=True)
     [... cut]