rerun-io/rerun #7411: `rerun_py/build.rs` leads to error in nixpkgs packaging

RobWalt, 2024-09-12:
I'm currently updating the nixpkgs package of the rerun app and we run the python sdk tests on every build. However I had to patch the `rerun_py/build.rs` file to get everything to run like this:

https://github.com/NixOS/nixpkgs/pull/334854/commits/ffa38763a06d1582bab22404b9f76616552ccac6#diff-16a717928b24120d28cdc8a3638f7383e24438a652fa6c5c9289652ab1ac6669

I'm not sure it that is an nix specific issue or if it is a bug generally present in the repo. I just wanted to report it here.

If I don't patch the path, then I get an error that the directory specified in the `build.rs` doesn't exist, which is true. Not sure if something within your internal build system fixes this automatically or not.

jleibs, 2024-09-12:
@RobWalt sorry for the confusion on that -- sounds like we definitely need to write a `PACKAGING.md` doc to help folks maintaining downstream packages like this.  Also, thanks for helping out with the nixpkgs -- this has been a surprisingly popular distribution channel for us.

Our `rerun-sdk` python package actually inlines a build of the Rerun CLI executable into the python package so that when you pip install it you end up with a runnable standalone viewer.

If you want to replicate this functionality, the nix build script should likely be doing something similar to what we do in the conda package:
- https://github.com/conda-forge/rerun-sdk-feedstock/blob/main/recipe/build.sh#L38-L47

That said, if nix is packaging the Rerun-CLI package separately, we should probably just leave it out of the python package all together.  I think you can can achieve this by simply setting `RERUN_ALLOW_MISSING_BIN=1`, which I believe should be easier than the patching the `build.rs` file as you have done.

jleibs, 2024-09-12:
I created an issue for the meta-problem of documenting and tracking this stuff here:
 - https://github.com/rerun-io/rerun/issues/7413

RobWalt, 2024-11-27:
It looks like the patch isn't needed anymore in the latest release

https://github.com/NixOS/nixpkgs/pull/358093

closed 2024-11-27 by RobWalt