rerun-io/rerun #7412: Have background stripes also in expanded rows in the dataframe view

abey79, 2024-09-12:
We should have background stripes everywhere. This means we must compute the odd-ness for rows (and sub-rows) on the fly, based on the expansion cache.

emilk, 2024-09-22:
I'm not sure that expanding one row into multiple lines should change the background stripes of existing rows. That might just be confusing. Maybe it is better to sometimes have double-stripes, or stripe expanded rows differently

abey79, 2024-09-23:
Actually, that has already landed with #7066 (so I'm going to close this issue).

I had the exact same thought when implementing that, but I noticed because everything shifts up and down when rows are expanded/collapsed, it's _extremely_ difficult to notice the "stripe inversion" on subsequent rows. So we're pretty good as is imo.