rerun-io/rerun #7417: Variants for entity fields

aecins, 2024-09-13:
**Is your feature request related to a problem? Please describe.**
I would like to be able to visualize a pointcloud in ReRun with different colorization options. For example we may want to colorize a pointcloud based on `DISTANCE_FROM_SENSOR`, `SURFACE_NORMAL_ORIENTATION`, `SEMSEG_CLASS`. ReRun viewer GUI should allow switching between the different colorization options. This feature would make ReRun a more powerful tool for visualizing 3D reconstruction results or debugging robotics 3D perception outputs. 

**Describe the solution you'd like**
A possible solution is to allow entities to have variants for its fields. For the example above, a `Points3D` entity could have multiple variants for the `colors` field. The blueprint is then  responsible for deciding which one of the variants is currently visualized.

**Describe alternatives you've considered**
The desired behavior can be "approximated" with the current build by logging multiple `Point3D` entities with identical point positions but different point colors. However, this approach has the following downsides:
1. it wastes memory space by logging duplicated point position data
2. it creates multiple separate entities for the same pointcloud. This means that changing the colorization mode requires at least two toggle switches: one to turn off current colorization and one to turn on the desired colorization. Also, it is not clear what the behavior of the visualizer should be if multiple pointcloud entities with the same point positions are visible at the same time.

**Additional context**

jpazhaya, 2024-09-18:
Colorizing options within the visualizer would be great too. i.e. just like how rviz enables coloring point clouds by selecting channel and color scales.

Wumpf, 2024-09-18:
@jpazhaya btw. there's already some limited edit options for colors and other things in the viewer via the override panel when you select an entity :)
I imagine predefined color sets could be part of this ask for sets (predefined ones included?) to choose from!

See also
* https://github.com/rerun-io/rerun/issues/6897
* https://github.com/rerun-io/rerun/issues/3860
... and I'm pretty sure there was something about "color derived from other values" (like velocity) but can't find it right now 🤔

teh-cmc, 2024-09-23:
That's also very much in line with tagged components: one could log a bunch of Color components with user-defined tags, then switch the tag in use at runtime using the UI and/or blueprint APIs (i.e. visualizer overrides).
* #6889

adhityabg, 2025-04-11:
This feature will be extremely useful for me and my usecase as well

closed 2026-02-12 by Wumpf