rerun-io/rerun #7420: Can't show the video twice

Wumpf, 2024-09-16:
We added the ability to reference videos, but our video decoder goes crazy if it has to show frames from different locations at the same time (that's expected the way it's written). Therefore:

* Either remove video reference feature for now
* Or add a hashmap of "decode tracks": Each decode track is just another decoder using the same video data (arc)
    * identification via entity path (+ override entity path if there. Super important otherwise overrides will cause this issue again!)
    * aggressively GC when unused for a frame

I think we can implement that track mechanism easily enough. But if it turns out to be a hassle we can cut back!

emilk, 2024-09-23:
This should also be able to unlock playing back a video in the selection panel (preview selected video)

Wumpf, 2024-09-23:
I don't think that's a good idea universally, at least not the simple solution of just looping a video in there: High resolution videos may take a serious amount of processing & memory that we don't want to waste on a small preview. Also if it just keeps playing in there that might be highly distractive. So at least some controls would be needed.
I'd rather pick a still from the video (how?) and discard the decoder once possible

PR #7473 (2024-09-23): Allow showing the same videos at several timestamps at the same time
### What

* Fixes #7420

https://github.com/user-attachments/assets/5f0a1f76-fd56-4f80-87f9-c3e02cd6e8d2
Decoders are now lazily created. There can now be several per video, each identified by a user provided identifier.
If decoders are not used for a frame, they get automatically destroyed.

Also fixes flickering issue when switching from an video error state to pending - now whenever we enter the pending state after an error, we clear out the target texture first.

Commit-by-commit review enabled.

[image]

### Checklist
*  [... cut]
commit bbf7d95e (2024-09-23): Allow showing the same videos at several timestamps at the same time (#7473)

### What

* Fixes #7420

https://github.com/user-attachments/assets/5f0a1f76-fd56-4f80-87f9-c3e02cd6e8d2
Decoders are now lazily created. There can now be several per video,
each identified by a user provided identifier.
If decoders are not used for a frame, they get automatically destroyed.

Also fixes flickering issue when switching from an video error state to
pending - now whenever we enter the pending state after an error, we
clear out the target texture first.

Commit-by-commit review enabled.

![example
screen [... cut]
closed 2024-09-23 by Wumpf