rerun-io/rerun #7439: Fix component reflection such as to not force the component to `impl Default`

abey79, 2024-09-18:
Currently, the codegen'd component reflection always emit a call to `C::default()`, forcing the component to `impl Default`. Sometime this really is meaning less (e.g what's a default `EntityPath` or `ComponentName`?).

Funnily the `ComponentReflection` struct actually has an `Option`. The codegen should therefore emit `None` if the component doesn't have `Default` (which the codegen knows from the `attr.rust.derive` attribute).

kpreid, 2024-09-18:
I ran into this myself when [adding `ShowLabels`](https://github.com/rerun-io/rerun/pull/7249/). `ShowLabels` is a component whose fallback value [is _always_ computed from other data,](https://github.com/rerun-io/rerun/blob/6121840ba46528c727197bc71d826a5e9527e869/crates/viewer/re_space_view_spatial/src/visualizers/utilities/labels.rs#L85-L110) and does not have a default that is meaningful at all.

Wumpf, 2024-09-18:
We discussed this internally quite a bit internally
https://rerunio.slack.com/archives/C041NHU952S/p1726669945999429

The gist of it that we do need guaranteed defaults for anything that's potentially ui editable (for the value to start out with). Overall, the force to default was introduced as a simplification so that anyone querying a fallback can rely on it being there and treat it as an hard error when it isn't.
It is a known issue that `Default` trait us such is slightly misleading semantically here. When we introduced this we discussed the possibility of a separate trait but discarded this because of the added overhead.

Wumpf, 2024-09-18:
@kpreid `ShowLabels` would be a case where a default value _is_ important. The reason is that one can edit it in the visualizer ui and the ui needs a value to start with.

Wumpf, 2024-09-18:
Have some work in progress that allows to generate placeholders fully generically so we need lot less of those pointless looking defaults and can _actually_ make use of the fact there's _always_ a placeholder 🙂 
https://github.com/rerun-io/rerun/tree/andreas/automatic-component-placeholders

since we still want to use `Default` for custom placeholders when available I think we'll have to make it opt-out as proposed in https://github.com/rerun-io/rerun/pull/7440
.. unless someone does a codegen thing where we check whether `Default` procmacro is used or there's a `Default` impl in the extension :)

closed 2024-10-25 by Wumpf