rerun-io/rerun #7445: Fail to cross compile arrow_cpp in rerun_cpp_sdk

SunDoge, 2024-09-19:
**Describe the bug**

Fail to cross compile rerun_cpp_sdk to `aarch64`. 

In `arrow_cpp-configure-out.log`

```
-- System processor: x86_64
-- Performing Test CXX_SUPPORTS_SSE4_2
-- Performing Test CXX_SUPPORTS_SSE4_2 - Failed
-- Performing Test CXX_SUPPORTS_AVX2
-- Performing Test CXX_SUPPORTS_AVX2 - Failed
-- Performing Test CXX_SUPPORTS_AVX512
-- Performing Test CXX_SUPPORTS_AVX512 - Failed
-- Arrow build warning level: PRODUCTION
-- Configuring incomplete, errors occurred!
```
And `arrow_cpp-configure-err.log`

```
CMake Error at cmake_modules/SetupCxxFlags.cmake:472 (message):
  SSE4.2 required but compiler doesn't support it.
Call Stack (most recent call first):
  CMakeLists.txt:415 (include)
```

**To Reproduce**

I am unable to provide reproduction steps as I am using an internal system.

**Expected behavior**

Can be fixed by adding `-DCMAKE_TOOLCHAIN_FILE=${CMAKE_TOOLCHAIN_FILE}` to `download_and_build_arrow.cmake`

```
-- System processor: aarch64
-- Performing Test CXX_SUPPORTS_SVE
-- Performing Test CXX_SUPPORTS_SVE - Success
-- Arrow build warning level: PRODUCTION
-- Using ld linker
-- Configured for RELEASE build (set with cmake -DCMAKE_BUILD_TYPE={release,debug,...})
-- Build Type: RELEASE
-- Performing Test CXX_LINKER_SUPPORTS_VERSION_SCRIPT
-- Performing Test CXX_LINKER_SUPPORTS_VERSION_SCRIPT - Success
```

**Screenshots**

**Backtrace**

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04.6 LTS x86_64

**Rerun version**

v0.17, v0.18

**Additional context**

Related issues:
https://github.com/apache/arrow/issues/30925

PrasanBora, 2024-09-19:
You should modify the download_and_build_arrow.cmake script to include the -DCMAKE_TOOLCHAIN_FILE=${CMAKE_TOOLCHAIN_FILE} option
This will ensure that CMake uses the correct settings for cross-compiling to aarch64

Wumpf, 2024-09-19:
Related to:
* https://github.com/rerun-io/rerun/issues/7445

our CMake files are generally not set up and tested for cross compilation right now. But please do try @PrasanBora 's suggestion. If it works a doc update PR would be in order :)

SunDoge, 2024-09-19:
> Related to:
> 
>     * [Fail to cross compile arrow_cpp in rerun_cpp_sdk #7445](https://github.com/rerun-io/rerun/issues/7445)
> 
> 
> our CMake files are generally not set up and tested for cross compilation right now. But please do try @PrasanBora 's suggestion. If it works a doc update PR would be in order :)

I have already tried it before and it works. I open this issue because I hope `-DCMAKE_TOOLCHAIN_FILE=${CMAKE_TOOLCHAIN_FILE}` can be added to the official repo so I don't have to patch it.

Wumpf, 2024-10-22:
@SunDoge if that's all there is to it, would you mind opening a PR? :)

SunDoge, 2024-10-22:
> @SunDoge if that's all there is to it, would you mind opening a PR? :)

Sure, I'll open a PR in the next few days.

PR #7866 (2024-10-23): Foward `CMAKE_TOOLCHAIN_FILE` to arrow build for sdk cross-compilation

### What

Add `-DCMAKE_TOOLCHAIN_FILE=${CMAKE_TOOLCHAIN_FILE}` to the cmake file to fix #7445 

### Checklist
* [x] I have read and agree to [Contributor Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and the [Code of Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested the web demo (if applicable):
  * Using examples from latest `main` build: [rerun.io/viewer](https://rerun.io/viewer/pr/7866?manifest_url=ht [... cut]
commit 246c1fba (2024-10-23): Foward `CMAKE_TOOLCHAIN_FILE` to arrow build for sdk cross-compilation (#7866)

### What

Add `-DCMAKE_TOOLCHAIN_FILE=${CMAKE_TOOLCHAIN_FILE}` to the cmake file
to fix #7445

### Checklist
* [x] I have read and agree to [Contributor
Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and
the [Code of
Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested the web demo (if applicable):
* Using examples from latest `main` build:
[rerun.io/viewer](https://rerun.io/viewer/pr/7866?manifest_u [... cut]
closed 2024-10-23 by Wumpf