rerun-io/rerun #7454: BGR Encoding doesn't work on web in all platforms

jleibs, 2024-09-19:
```
pixi run rerun-web
pixi run -e py python tests/python/release_checklist/check_bgr.py 
```

Browsers: Firefox and Chrome both show a black tile:
[image]

Works fine on native.

OS: Arch, Wayland, Nvidia (560 drivers)

jleibs, 2024-09-19:
Additional info from web console:
[image]

Wumpf, 2024-09-19:
Turns out webgl doesn't support BGR(A)
https://registry.khronos.org/OpenGL-Refpages/es2.0/xhtml/glTexImage2D.xml

I wonder how this fell through the cracks. I definitely tested Web when introducing this feature and I'm _fairly_ sure I had a look at WebGL _and_ WebGPU (where this works via spec). Might want to check if some WebGL impls allow this or (more likely) I only tested WebGPU.

Also there's a wgpu bug here where it doesn't set the downlevel flag for missing texture support.

Wumpf, 2024-09-19:
Somebody already logged this on wgpu :)
https://github.com/gfx-rs/wgpu/issues/3583

closed 2024-09-25 by Wumpf