rerun-io/rerun #7455: Dataframe API improvements

jleibs, 2024-09-19:
## Notes from exploration:
- EntityPathFilter expression is redundant with column-selection
- Different types of column filtering is very useful for exploration
- Select operators feel natural for incrementally filtering a dataset independently of the latest-at / range queries
  - Would be nice to just chain these in, or re-use a selection for multiple queries.
- Implicit empty columns can be surprising. Some filter operations should probably return errors if the columns are not present.
- Ultimately need a way to access deterministic/named columns in the context of the resulting Table
  - Allowing a user-set name in the `ComponentSelector`
- POV component is confusing
  - Really need a sampled multi-latest-at
  - Could be more clearly expressed as something like:
    ```
    let times = dataset.log_times(columns)
    dataframe = dataset.latest_at(times, columns)
    ```

nikolausWest, 2024-09-19:
Awesome writeup!

> The big challenge here is that sane-looking APIs are ambiguous without knowledge of the timeline.

One other possible direction to go here: What if you actually express the range or latest at query as an operation on the TimeColumnDescriptor? At that point you know what type it has in the store and can make it ergonomic. It's also symmetric in some way with how we handle other columns. It also avoids mistakes like doing `rr.dataframe.Range.seconds("time", min=10.0, max=17)` when "time" was in fact a poorly named sequence timeline.

jleibs, 2024-09-19:
> At that point you know what type it has in the store and can make it ergonomic.

I think it half-solves the problem, but it still doesn't actually handle `seconds` vs `nanoseconds` on a time-typed column, which is still its own problem. I think to do that we would need to introduce some kind of "natural units" metadata on the column, but that's also awkward and error prone.

I'm hesitant to pull in something like https://pypi.org/project/custom-literals/, but that's of course the kind of behavior that would really be nice.

nikolausWest, 2024-09-24:
Something I'm wondering about is how we handle multiple recording id's here.

Multiple rrds could all have the same recording is, so something like this makes sense:
```Python
recording = rr.data.load_recording("first.rrd", "second.rrd")
```
However, we can't know up front if those files contain one or two recording ids. How do we handle that?

- rr.data.load_recording returns a list
- rr.data.load_recording raises if there are multiple recordings
- rr.data.load_recording only takes a single file and you can merge them in the sdk instead.
- rr.data.laod_recording adds a dictionary encoded column with recording id

The same goes for application id and any future user defined ids

jleibs, 2024-09-26:
Reminder: we still maybe need an API for filtering all-empty columns (related to unused transform components).

closed 2024-10-11 by jleibs