rerun-io/rerun #7467: The datastore browser tables has weird alignment on the first non-header row

abey79, 2024-09-23:
[image]

abey79, 2024-09-23:
Somehow all rows are offset, but the first row seem more offset than the others:
[image]

abey79, 2024-09-23:
Table without zebra stripes look fine though:

[image]

emilk, 2024-10-11:
Does this affect the dataframe view?

abey79, 2024-10-11:
It doesn't seem to. The datastore browser is still on `egui_extras::Table` though, so not super surprising. Very much ok to punt on this for 0.19 imo.