rerun-io/rerun #7468: Timeline and `Clear` handling issues in dataframe range queries

abey79, 2024-09-23:
# Setup

Consider a relatively simple archetype with a `Clear` half-way through:

```python
def specimen_archetype_chunk_with_clear():
    rr.send_columns(
        "/archetype_chunk_with_clear",
        frame_times(range(10)),
        [
            rrc.Position2DBatch([(i, i) for i in range(10)]),
        ],
    )

    set_frame_time(0)
    rr.log_components("/archetype_chunk_with_clear", [rr.Points2D.indicator(), rrc.RadiusBatch(2)])

    set_frame_time(5)
    rr.log("/archetype_chunk_with_clear", rr.Clear(recursive=False))
```

[image]
[image]
[image]

# Expected

1) The initial radius=5 should be cleared from t=5
2) The 5th Position2D should be cleared (and then overwritten at t = 6).

Expectation (2) matches the visualiser behaviour (the point "disappear" at t=5 and reappear at t=6).

# Actual

Radius is indeed cleared, but the [5, 5] point is not.

[image]

teh-cmc, 2024-09-24:
The problem stems from the implementation of `JoinEncoding::OverlappingSlice`, specifically:
https://github.com/rerun-io/rerun/blob/05a8808852e0071a6ce0b9d37e00dd8ac98a000f/crates/store/re_dataframe/src/range.rs#L492-L505

This is only ever correct if A) the entity being queried has no `Clear` logged to it and B) none of its recursive parents have a recursive `Clear` logged to them.

If you switch the default `JoinEncoding` to a dictionary, the issue will go away since that path always fully resolves latest-at queries (and therefore accounts for clears):
```diff
diff --git a/crates/store/re_chunk_store/src/dataframe.rs b/crates/store/re_chunk_store/src/dataframe.rs
index 0a4b01e8f1..0c33cae1e9 100644
--- a/crates/store/re_chunk_store/src/dataframe.rs
+++ b/crates/store/re_chunk_store/src/dataframe.rs
@@ -37,7 +37,6 @@ pub enum JoinEncoding {
     /// This minimizes the need for allocation, but at the cost of `RecordBatch`es that are
     /// almost always smaller than the optimal size. In the common worst-case, this will result
     /// in single-row `RecordBatch`es.
-    #[default]
     OverlappingSlice,
 
     /// Dictionary-encode the joined column.
@@ -48,6 +47,7 @@ pub enum JoinEncoding {
     /// Note that this changes the physical type of the returned column.
     ///
     /// Using this encoding for complex types is incompatible with some arrow libraries.
+    #[default]
     DictionaryEncode,
     //
     // TODO(jleibs):
```
[image]

I don't fully understand all the implications of this code so I don't have any recommendations yet.

teh-cmc, 2024-10-09:
All good now:
[image]

closed 2024-10-10 by teh-cmc