rerun-io/rerun #7481: Mp4 crate/video types shouldn't copy large amounts of data

Wumpf, 2024-09-24:
Right now we effectively hold an mp4 blob at least twice: Raw in the (arrow) store and then again upon parsing in our mp4 object.
All larger pieces should just be pointers back to the raw blob in order to conserve memory.

Also general cleanup of mp4 crate is in order, this is a good opportunity.

emilk, 2024-09-24:
Does arrow bottom out in https://crates.io/crates/bytes ? That would be useful for storing RC-counted references across crates. (though I don't believe re_mp4 needs to own a reference to the bytes, do it?)

emilk, 2024-10-08:
I think the best path forward is to change `re_mp4` to store byte ranges (`Range<usize>`) rather than the raw bytes, and then have `re_video` (or `re_renderer`) have some sort of `trait ByteSource { fn read(&self, bytes: Range<usize>) -> Option<Vec<u8>> }` that we then can wrap around a `Blob`.

It should be less than a days worth of work.

jprochazk, 2024-10-22:
Even after https://github.com/rerun-io/rerun/pull/7860 we still copy video data for each sample. But `Blob` is a `Arc<Vec<u8>>` internally, so we should be able to avoid that as well. The decoder could receive a sub-slice of the Blob instead of copying the data to its own `Vec`. For web, it's possible to give the JS video APIs a direct reference to Wasm's linear memory, using the unsafe [`Uint8Array::view`](https://rustwasm.github.io/wasm-bindgen/api/js_sys/struct.Uint8Array.html#method.view), which means constructing the `EncodedVideoChunk` will not require a copy either.

jprochazk, 2024-10-23:
Follow-up: https://github.com/rerun-io/rerun/issues/7878

closed 2024-10-23 by jprochazk