rerun-io/rerun #7484: Support for streaming encoded video from continuous log/send calls

Wumpf, 2024-09-24:
We want to be able to support streaming video in via continuous log/send calls.

The rough sketch for this feature would be to introduce an `VideoStream` archetype to which users continuously add `VideoSample` components.
The `VideoStream` archetype has some meta data that is ideally logged as static (so it's never GC'ed even if we lose samples).
Each `VideoSample` component is equivalent to a sample in our decoder abstraction - so ideally we just do a `bytemuck` cast to the decoder and no further processing is needed.
All samples on the `VideoStream`s are fed to the decoder at once, we essentially ignore the timeline since necessarily all samples come with their own decode & presentation timestamp.
Constraint: samples have to come in decode timestamp order (otherwise we have to do reordering which is expensive perf & memory wise for long streams; also would break under GC)

Open questions: 
* what does a typical workflow for this look like
   * obviously we have to provide samples where we encode a video stream e.g. with gstreamer and send it continuously to the viewer
* Exact mechanics of sample logging 
    * where on the timeline are samples added?
    * is the timeline really completely (other than ordering) and how do we communicate this? Should we encourage index timeline which is independent of the timeline that contains the `VideoFrameReference`, etc.
    * are these arrays of samples on each timeline step? (why not?)
* can we support b-frames? Most likely, but there might be jankiness if provided incorrectly

emilk, 2024-10-11:
## Workaround
If you can't wait for us to implement this feature, you can work around it by logging your video stream as many shorts videos, each logged as a `AssetVideo` to the same Entity Path.

patrickelectric, 2025-05-20:
This is a really important feature!

nikolausWest, 2025-06-17:
@patrickelectric this feature is on it's way now. Do you think you could share more about your use case and need here. What larger problem are you trying to solve? Which specific parts are most important to you? Can you think of a single demo that would communicate to you that this feature solves your needs?

patrickelectric, 2025-06-17:
Hi @nikolausWest thanks for the questions.

I work with @bluerobotics developing [BlueOS](https://github.com/bluerobotics/BlueOS) for [ROVs](https://en.wikipedia.org/wiki/Remotely_operated_underwater_vehicle) and [USVs](https://en.wikipedia.org/wiki/Unmanned_surface_vehicle) (that can also be used for aerial drones). We are working towards [zenoh](https://github.com/eclipse-zenoh/zenoh) integration in our system and using foxglove messages and at the moment integrating mcap recording.

With that said... We already have [cockpit](https://github.com/bluerobotics/cockpit) for vehicle control and some data visualization. But we want to use rerun to a more in depth approach, where we can see sonar, lidars, video, log and have everything synchronized to debug logs and visualize real-time vehicle data. Low latency, or a glass-to-glass latency of 200ms or less (100ms) would be awesome. 

A simple and ideal demo would be a gstreamer pipeline or a raw h264 video file publishing data to rerun to visualize the stream.

Wumpf, 2025-06-17:
> A simple and ideal demo would be a gstreamer pipeline or a raw h264 video file publishing data to rerun to visualize the stream.

I didn't have much success with gstreamer so far mostly because of setup issues on Mac. But I got examples for live streaming h264 video from ffmpeg on the way. So sounds like this gonna be a great fit :)
Depending on platform and amount of data live decoding in the viewer still struggles a little bit, but in the web viewer things are already quite smooth. We'll likely ship more improvements in 0.25 towards that.

I'll post a link of the example here soon; would be awesome if you can have a look or even try a nightly build with the feature :)

Wumpf, 2025-06-27:
We got a maintained snippet and an example now for the new `VideoStream` archetype!
* https://github.com/rerun-io/rerun/blob/f8a3c2408ba4bd5a4af8ac659e8f19f92c264ba7/docs/snippets/all/archetypes/video_stream_synthetic.py
* https://github.com/rerun-io/rerun/tree/f8a3c2408ba4bd5a4af8ac659e8f19f92c264ba7/examples/python/camera_video_stream

Any feedback would be welcome very much, otherwise we'll most likely ship with this interface.

The only thing missing for closing this ticket is updated video reference docs.
There are also some smaller improvements (and regressions) on video in general that I want to land for 0.24, but nothing crazy

closed 2025-06-30 by Wumpf

cmpute, 2025-12-27:
With this function landed, it seems that CompressedVideo messages in mcap files are still not supported directly. Is there any tutorial on visualizing mcap files with foxglove's CompressedVideo messages? Are we expected to convert the message to rerun's native VideoStream before visualization?