rerun-io/rerun #7493: Tombstone data should not be returned by default in high-level dataframe APIs

teh-cmc, 2024-09-24:
Their semantics always apply no matter what -- a tombstone is a tombstone -- but users don't want to look at the tombstone themselves in 99% cases.

abey79, 2024-09-24:
Yes.

In terms of Jeremy's new proposal, to me it make sense to see the tombstone in the "raw" data (e.g. where each row corresponds to a log event), because at this point, the tombstone semantic is not yet applied. But as soon as it is applied, it should no longer be visible.

jleibs, 2024-10-04:
I realized we didn't specify what to do with tombstones as part of https://www.notion.so/rerunio/Repeated-Time-Values-10bb24554b1980d8a0cdcca3d8903502

I agree generally hiding them from the view makes sense for now.

closed 2024-10-10 by teh-cmc