rerun-io/rerun #7526: Configure UI to step by time rather than event

jleibs, 2024-09-26:
For very dense timeline it's nice to have a granularity for "stepping forward" that is based on time delta, e.g. "0.1 sec" rather than always stepping to the next event.

This would need to be configurable, since there is no universal granularity that makes sense.

andybarry, 2024-09-27:
Thanks for adding this!  I'm the one who asked about this on the discord.  What I'd really like is another hotkey for it since the arrows keys are definitely useful!  I can tell you all the ones I guessed at in hopes of it working, which are perhaps good possibilities:
- Shift+arrows
- Alt+arrows
- Left and right brackets `[` `]`
- Comma and period `,` `.` because they have the `<` and `>` keys on them and are used in YouTube and other apps for speedup/slowdown.  (Also tried these with shift pressed)

In other apps I've used, 0.1 seconds and 1.0 seconds are good defaults, I tend to like having both.  If I had my choice I'd do:
- `Alt+arrows` for 0.1 seconds forwards and back.
- `Shift+arrows` for 1.0 seconds forwards and back

Wumpf, 2025-12-17:
on `main`  / in upcoming 0.28 we have ➡️ / ⬅️ for 0.1 steps and shift + ➡️ / ⬅️ for 1s steps