rerun-io/rerun #7607: Native video decoder via `ffmpeg` CLI

emilk, 2024-10-07:
Send samples to `ffmpeg` over CLI, read back the results, show the user.

TODO: fill this in with more detals

Wumpf, 2024-11-04:
Left after the initial PR:

* [ ] test with broken samples (completely invalid chunks)
* [ ] download ffmpeg for the user or at least tell users what to do when there's no ffmpeg (haven't checked even yet on the error message)
* [ ] add version & ability check
* [ ] consume 420p and friends directly to speed up processing

Wumpf, 2024-11-05:
process startup isn't much of an issue according to profiler, but wait for shutdown can cause severe hickups:
[image]

closed 2024-11-06 by Wumpf