rerun-io/rerun #7608: Unify GPU texture conversion (with focus on chroma downsampled formats)

Wumpf, 2024-10-07:
Today, we have a variety of formats that we directly support by out "rectangle" shader.
There's a few issues with this though:
* every time we make it longer we see issues on WebGL on Linux
* we keep appending to the shader that gets more and more complicated https://github.com/rerun-io/rerun/issues/3931

TODO: more words

emilk, 2024-10-07:
The supported pixel formats should map 1-to-1 to `rerun::PixelFormat`.

We should consider following the naming convention of https://facebookresearch.github.io/ocean/docs/images/pixel_formats_and_plane_layout/

emilk, 2024-10-07:
Here are the pixel formats we need to support for video: https://docs.rs/dav1d/0.10.3/dav1d/enum.PixelLayout.html

Wumpf, 2024-10-07:
And these are the color primaries we have to support:
https://docs.rs/dav1d/0.10.3/dav1d/pixel/enum.ColorPrimaries.html

For the moment we shouldn't try to support everything and focus on the most common ones. Which is likely just `BT709` for now. I found BT2020 source material as well, but this gets us into HDR support territory, see
* #7594

Wumpf, 2024-10-11:
We have what we needed for video. Closing this in favor of this follow-up task:
* https://github.com/rerun-io/rerun/issues/7700

closed 2024-10-11 by Wumpf