rerun-io/rerun #7619: Rename `NV12` -> `Y_UV12`, `YUY2` -> `YUYV16`

Wumpf, 2024-10-07:
We want to start following the naming convention established by the Ocean library for naming chroma downsampling formats. In particular this is a lot better at distinguishing interleaved, planer & semi-planar formats!

https://facebookresearch.github.io/ocean/docs/images/pixel_formats_and_plane_layout/

emilk, 2024-10-07:
Here are the pixel formats we need to support for video: https://docs.rs/dav1d/0.10.3/dav1d/enum.PixelLayout.html

Wumpf, 2024-10-09:
Slight change of mind: The problem with ocena's denomination is that it also encodes bit depth into the same breath. I don't think we want to do this. At least in `re_renderer` land what already makes much more sense is to denominate the standard [downsampling ratios](https://en.wikipedia.org/wiki/Chroma_subsampling#Sampling_systems_and_ratios) (even though they are insane imho🤦 ) instead of the bit depth

E.g. instead of `Y_U_V24` use `Y_U_V_444` 

cc: @jleibs