rerun-io/rerun #7626: Textlog does not support chinese

haixuanTao, 2024-10-08:
Hello,

`Textlog` does not support chinese character.

[image]

Is there a way to make it work? 

Thanks

emilk, 2024-10-08:
Adding support for Chinese should indeed be possible, but the way our GUI library is set up (egui) it would mean bundling a Chinese font in the binary, or building out egui to load system fonts when needed.

Bundling a Chinese font is definitely the easier route, but it means each wasm binary gets bigger by quite a lot, so I think I rather figure out a way to get egui to dynamically load fonts when needed:
* https://github.com/emilk/egui/issues/5233

haixuanTao, 2024-10-08:
That make a lot of sense. Thanks for the quick response.

I think that I can alleviate this problem for now but, I think it could be nice to have. Also, I can imagine that chinese rerun settings can be a future feature as well. 

Working more and more on multimodal data and rerun is really nice for this !

Thanks :)

haixuanTao, 2024-10-08:
Closed in favor of #5234

emilk, 2024-10-08:
Let's keep this issue open to track the progress