rerun-io/rerun #7647: video testing: make sure we present the right frame at the right time

emilk, 2024-10-09:
We currently use latest-at semantics for selecting the current video frame.

However, this is prone to rounding errors.

We should have some sort of test that we are indeed presenting exactly the right frame when the user selects an exact time point.

We could consider adding a small bias to avoid the rounding errors, or switch to closest-at. Neither solution is very attractive.

emilk, 2024-10-20:
This seems to be a problem on Safari, and maybe elsewhere too: https://x.com/chrisoffner3d/status/1847625953544966249

Wumpf, 2024-10-31:
Acquired much better understanding of this by now, giving rise to two new bugs:
* https://github.com/rerun-io/rerun/issues/7961
* https://github.com/rerun-io/rerun/issues/7960