rerun-io/rerun #7671: re_viewer no longer compiles: dav1d_sgr_x_by_x' can not be used when making a shared object; recompile with -fPIC

jleibs, 2024-10-09:
Affected platforms: Arch Linux
- Regression caused by: https://github.com/rerun-io/rerun/pull/7661

jleibs, 2024-10-09:
Also fails in CI:
- https://github.com/rerun-io/rerun/actions/runs/11261410122/job/31314795062#step:6:386

jprochazk, 2024-10-10:
I can repro this. I tried setting `-Clink-args=-fPIC`, but that didn't help, it still claims it should be recompiled with `-fPIC`.

emilk, 2024-10-10:
So maybe some clash of dependencies then?

teh-cmc, 2024-10-10:
TL;DR:
* Only x64 platforms go through the `nasm`-based path, everything else uses the system's default assembler (I'm not sure why, but naively trying to use the system's assembler for x64 fails miserably).
* On Linux x64, the `nasm` path generates relocation entries that fail to resolve at link-time for some unknown reason, even though the whole thing is fully statically link, only use relative relocs, etc. Even weirder, it only fails when linking the full viewer, simpler examples do link successfully (it's possible that the problematic symbols are unused and therefore optimized out before the reloc pass for these examples?).
* On Mac x64, the `nasm` path just works. Somehow.

Wumpf, 2024-10-11:
On windows x64 it works like a charm. Given that it makes 8k video usable (completely unusable without), I think as a first step we should limit the changes of https://github.com/rerun-io/rerun/pull/7675 to Linux

Wumpf, 2024-10-11:
:sigh: that is ofc running against https://github.com/rust-lang/cargo/issues/1197, it's very hard to make this feature have only an effect on certain platforms

emilk, 2024-10-11:
I think we can work-around it by adding this to a crate _other_ than re_video

``` toml
[target.'cfg( not(target_os = "linux") )'.dependencies] # TODO(#7671)
re_video = { workspace = true, features = [ "nasm" ] }
```

emilk, 2024-10-15:
Should have been fixed by https://github.com/rerun-io/re_rav1d/pull/3, but lets test it before closing (and maybe follow-up and remove `#7671`-TODOs in the Rerun codebase)

emilk, 2024-10-15:
Apparently the `-F nasm` feature makes a 25x perf difference for @teh-cmc on Linux 64.

Based on that, I think we should disable av1 decoding without `-F nasm` on Linux and Windows (Mac is fast even without it).

closed 2024-10-15 by emilk