rerun-io/rerun #7678: Slow rendering causes data ingestion to be even slower

kpreid, 2024-10-10:
Problem: Suppose that

* you have a large amount of data to load, and
* that data is also slow to render — let's say 200 ms per frame.

In this case, the data will be extremely slow to load — but faster if you switch to a different recording that isn't so slow to draw. The culprit, I would guess, is this logic in `re_viewer`’s message receiving loop:

https://github.com/rerun-io/rerun/blob/06cfa9eb1c8facb435bf7de6991aa0513781e8a6/crates/viewer/re_viewer/src/app.rs#L1192-L1195

This means that after each 10 ms portion of the data is loaded, the viewer will redraw all the (visible) data loaded so far — which means in my example of 200 ms drawing time, means it's spending 95% of its time redrawing partially-loaded data instead of getting more data loaded, and over the entire process adds up to a total load time that _could_ be as bad as O(N²) (in the worst case where the redraw cost is proportional to the data loaded so far). In the actual scenario that inspired this report, my user experience is like: run application for 2 seconds, exit application, wait 1 minute for the viewer to finish thinking about the data it got.

Possible solutions:

* Keep the 10 ms interruptions but *not repaint* until a longer period has passed; this would require being able to `receive_messages()` without doing it inside `update()`.
* Less refactoring, less good: adjust the 10 ms limit upward whenever drawing is especially slow, which will reduce responsiveness but perhaps still be better on net.
* Ultimate engineering-heavy solution: concurrent `EntityDb` that allows the data source to be appending while the renderer is reading.

(Obviously, it would be better if the data wasn’t taking 200 ms to draw in this scenario. But, no matter how efficient the renderer gets, users will always have reasons to push the limits of what is feasible on the hardware they have, so I think it’s worth doing something here.)

chengguizi, 2024-12-26:
Hi Rerun team,

I have a possibly related observation / issue.

We have multiple streams of jpeg image data logged in real time. On some of our Intel laptop machines, when we unplugged the battery, the render time becomes longer (example, 45ms vs 15ms). When it is unplugged, the latency slowly piles up.

Visually it makes the viewing lagged the realtime more and more over time, and also memory usage grows indefinitely.

Thoughts:
- Can we control about the frame render behaviour. Basically, let the render to skip frames if the newer data has arrived, before the next frame to render is to kick start
- This will allow renderer to skip frames and keep to realtime as to the logging.

chengguizi, 2025-04-03:
Hi Rerun team,

Would like to bump this up to see if something could be planned upon! Appreciate :)