rerun-io/rerun #7729: Clearing entities with transforms leaves an identity transform behind

stonier, 2024-10-14:
**Describe the bug**

Everything seems to be working as documented, i.e.

* [Clear](https://docs.rs/rerun/latest/rerun/archetypes/struct.Clear.html) - empties all components of an entity
* [Spaces & Transforms](https://rerun.io/docs/concepts/spaces-and-transforms) - unless otherwise specified, every path is trivially connected to it's parent by the identity transform

However, it's the combination of the two when a non-identity transform is logged to an entity. It gets first cleared, and then reactivated as the identity keyframe which is subsequently visualized at the origin. 

**To Reproduce**

1. Log a transform under an entity name, e.g. `recording_stream.log("/bob", &transform);`
  (transform must not be the identity)
2. Clear the entity, e.g. `recording_stream.log("\bob", &rerun::Clear::recursive());
3. View the replay - the original transform keyframe is replaced by an identity transform keyframe at the origin

**Expected behavior**

The transform is not restored (and thus not re-visualized).

**Desktop (please complete the following information):**
 - OS: Ubuntu 24.04

**Rerun version**
0.18.2

stonier, 2024-10-15:
A reproduceable program:

```rust
use rerun as rr;

fn generate_transform(x: f32) -> rr::Transform3D {
    rr::Transform3D::from_translation_rotation(
        rr::Vec3D::new(x, 0.0, 0.0),
        rr::Quaternion::IDENTITY,
    )
}

fn main() -> Result<(), Box<dyn std::error::Error>> {
    let (stream, storage) = rr::RecordingStreamBuilder::new("Clear").memory()?;
    let t_3 = generate_transform(3.0);
    let t_5 = generate_transform(5.0);
    let t_6 = generate_transform(6.0);
    stream.set_time_seconds("sim_time", 0.0);
    stream.log("/foo", &t_3)?;
    stream.log("/bar", &t_5)?;
    stream.set_time_seconds("sim_time", 1.0);
    stream.log("/foo", &t_3)?;
    stream.log("/bar", &t_6)?;
    stream.set_time_seconds("sim_time", 2.0);
    stream.log("/foo", &t_3)?;
    stream.log("/bar", &rerun::Clear::flat())?;
    rr::native_viewer::show(storage.take())?;
    Ok(())
}
```

Nikita240, 2024-10-15:
Here is a workaround for now:

```rust
use rerun as rr;

fn generate_transform(x: f32) -> rr::Transform3D {
    rr::Transform3D::from_translation_rotation(
        rr::Vec3D::new(x, 0.0, 0.0),
        rr::Quaternion::IDENTITY,
    )
}

fn main() -> Result<(), Box<dyn std::error::Error>> {
    let (stream, storage) = rr::RecordingStreamBuilder::new("Clear").memory()?;
    let t_3 = generate_transform(3.0);
    let t_5 = generate_transform(5.0);
    let t_6 = generate_transform(6.0);
    stream.set_time_seconds("sim_time", 0.0);
    stream.log("/foo", &t_3)?;
    stream.log("/bar", &t_5)?;
    stream.set_time_seconds("sim_time", 1.0);
    stream.log("/foo", &t_3)?;
    stream.log("/bar", &t_6)?;
    stream.set_time_seconds("sim_time", 2.0);
    stream.log("/foo", &t_3)?;
    stream.log("/bar", &rerun::Clear::flat())?; // <-- sends the visual 3d axes back to the origin
    stream.log("/bar", &rr::Transform3D::clear().with_axis_length(0.0))?; // <-- workaround
    rr::native_viewer::show(storage.take())?;
    Ok(())
}
```

stonier, 2024-10-15:
Problems with the workaround:

Step 1: Use that zero'd axis length to make them vanish (easy)

_but ... if entities are far from the origin, rerun zooms out excessively to try and fit in a transform that has been sent to the origin_

Step 2: Reset the transform with the last transform (more of a pain)

_but ... none of this handles transforms that have been cleared for sub-entities when using recursive_

Step 3: Reset the transforms of all subentities with their last transforms (can be too difficult to be worth chasing)