rerun-io/rerun #7766: `rr.serve` should be `rr.serve_websocket`, `rr.connect` should be `rr.connect_tcp`

teh-cmc, 2024-10-16:
And similarly, `rerun --serve` should be `rerun --serve-websocket` (or something along these lines, you get the idea).

Users get understandably confused by this. They expect a serving process (`rr.serve()`) to be ready to accept connections from other processes (`rr.connect()`), except the two things are in reality completely unrelated: one is hosting a websocket server to be polled by the web-viewer, while the other is trying to connect to the TCP sdk comms pipeline. 

See e.g. this thread: https://discord.com/channels/1062300748202921994/1295914647165210634

zehiko, 2024-10-16:
+1, I ran into this confusion myself just yesterday

teh-cmc, 2024-10-25:
Yet another thing that would make documenting the app model (#919) easier.

emilk, 2024-10-25:
Let's just rename these and keep the old names as deprecated redirects

PR #7906 (2024-10-28): `rr.serve` -> `rr.serve_web`, `rr.connect` -> `rr.connect_tcp`
Title.

* Fixes #7766 

* [x] :sheep:
closed 2024-10-28 by teh-cmc