rerun-io/rerun #7791: `rr.log_file_from_path` just hangs indefinitely

teh-cmc, 2024-10-17:
Try this:
```python
import sys
import rerun as rr

path_to_rrd = sys.argv[1]
rr.init("rerun_example_dataframe", spawn=True)
rr.log_file_from_path(path_to_rrd)
```

On 0.18.2, this works just fine.
On latest alpha, this does the logging as expected, but the hangs indefinitely. I literally cannot kill the logging process without resorting to the nuclear tools.

teh-cmc, 2024-10-17:
This seems to be only true with rrd files, passing e.g. a jpeg to that program seems fine.

emilk, 2024-10-17:
Likely caused by:
* https://github.com/rerun-io/rerun/pull/7475

teh-cmc, 2024-10-17:
Oh interesting, I see...

emilk, 2024-10-17:
Seems like there are two problems:
A) why doesn't the above .rrd have an `end-of-stream` marker?
B) why doesn't ctrl-C work?

teh-cmc, 2024-10-17:
A) The file was saved from the viewer using the `File > Save` feature -- it seems this does not add the proper EOS marker
B) I don't know how this is implemented yet, but I assume this is constantly retrying to `read()`? which would eat the signal

teh-cmc, 2024-10-17:
Okay I'll fix A to start with, that should be trivial.

teh-cmc, 2024-10-17:
As for B), looking at the core dump it's relatively safe to assume that's what's happening in the RRD dataloader hangs forever, and therefore the dataloader manager is stuck waiting for its thread:
```
#2  0x000070b8f3253b91 in std::sys::pal::unix::thread::Thread::join () from /home/cmc/dev/rerun-io/rerun/rerun_py/rerun_bindings/rerun_bindings.abi3.so
#3  0x000070b8f2de5b22 in std::thread::JoinInner<T>::join () from /home/cmc/dev/rerun-io/rerun/rerun_py/rerun_bindings/rerun_bindings.abi3.so
#4  0x000070b8f2dd68d9 in re_sdk::recording_stream::RecordingStreamInner::wait_for_dataloaders () from /home/cmc/dev/rerun-io/rerun/rerun_py/rerun_bindings/rerun_bindings.abi3.so
#5  0x000070b8f2ddd9bf in re_sdk::recording_stream::RecordingStream::disconnect () from /home/cmc/dev/rerun-io/rerun/rerun_py/rerun_bindings/rerun_bindings.abi3.so
#6  0x000070b8f2be531d in pyo3::marker::Python::allow_threads () from /home/cmc/dev/rerun-io/rerun/rerun_py/rerun_bindings/rerun_bindings.abi3.so
#7  0x000070b8f2c63287 in rerun_bindings::python_bridge::__pyfunction_shutdown () from /home/cmc/dev/rerun-io/rerun/rerun_py/rerun_bindings/rerun_bindings.abi3.so
```

closed 2024-10-17 by teh-cmc