rerun-io/rerun #7802: Non-linearity in dataframe read performance

jleibs, 2024-10-17:
Each doubling of the rows increases the read-time by a bit less than 4x.

Example:
```python
import rerun as rr
import tempfile
import time
import numpy as np

rr.init("rerun_example_many_scalars")

recordings = []

with tempfile.TemporaryDirectory() as tmpdir:
    for sz in [1, 20000, 40000, 80000, 160000, 320000]:
        rrd = tmpdir + f"/tmp_{sz}.rrd"
        rr.save(rrd)

        times = np.arange(0, sz)
        big_scalars = rr.components.ScalarBatch(np.sin(times))

        rr.send_columns(
            "signal",
            [rr.TimeSequenceColumn("time", times)],
            [big_scalars],
        )

        # Wait for flushing
        time.sleep(1)

        recordings.append(rr.dataframe.load_recording(rrd))

for recording in recordings:
    start = time.time()
    table = recording.view(index="time", contents="/**").select().read_all()
    print(f"Read: {len(table)} rows in {time.time() - start:.2f} seconds")
```

```
Read: 1 rows and in 0.00 seconds
Read: 20000 rows and in 0.37 seconds
Read: 40000 rows and in 1.25 seconds
Read: 80000 rows and in 4.27 seconds
Read: 100000 rows and in 6.55 seconds
```

emilk, 2024-10-17:
Accidentally quadratic then. I would consider that a performance bug.

teh-cmc, 2024-10-17:
Rust repro:

Run this:
```
import rerun as rr
import tempfile
import time
import numpy as np

rr.init("rerun_example_many_scalars")

recordings = []

for sz in [1, 20000, 40000, 80000, 160000, 320000, 640000]:
    rrd =  f"/tmp/tmp_{sz}.rrd"
    rr.save(rrd)

    times = np.arange(0, sz)
    big_scalars = rr.components.ScalarBatch(np.sin(times))

    rr.send_columns(
        "signal",
        [rr.TimeSequenceColumn("time", times)],
        [big_scalars],
    )

    # Wait for flushing
    time.sleep(1)

    recordings.append(rr.dataframe.load_recording(rrd))
```

Apply this patch:
```diff
diff --git a/crates/store/re_dataframe/examples/query.rs b/crates/store/re_dataframe/examples/query.rs
index 7d374587d4..2907e3f666 100644
--- a/crates/store/re_dataframe/examples/query.rs
+++ b/crates/store/re_dataframe/examples/query.rs
@@ -24,23 +24,8 @@ fn main() -> anyhow::Result<()> {
     };
 
     let path_to_rrd = get_arg(1);
-    let timeline_name = args.get(2).map_or("log_time", |s| s.as_str());
-    let time_from = args.get(3).map_or(TimeInt::MIN, |s| {
-        TimeInt::new_temporal(s.parse::<i64>().unwrap())
-    });
-    let time_to = args.get(4).map_or(TimeInt::MAX, |s| {
-        TimeInt::new_temporal(s.parse::<i64>().unwrap())
-    });
-    let entity_path_filter = EntityPathFilter::try_from(args.get(5).map_or("/**", |s| s.as_str()))?;
 
-    // TODO(cmc): We need to take a selector, not a Timeline.
-    let timeline = match timeline_name {
-        "log_time" => Timeline::new_temporal(timeline_name),
-        "log_tick" => Timeline::new_sequence(timeline_name),
-        "frame" => Timeline::new_sequence(timeline_name),
-        "frame_nr" => Timeline::new_sequence(timeline_name),
-        _ => Timeline::new_temporal(timeline_name),
-    };
+    let timeline = Timeline::new_sequence("time");
 
     let stores = ChunkStore::from_rrd_filepath(
         &ChunkStoreConfig::DEFAULT,
@@ -73,11 +58,13 @@ fn main() -> anyhow::Result<()> {
         };
         eprintln!("{query:#?}:");
 
+        let now = std::time::Instant::now();
         let query_handle = query_engine.query(query.clone());
-        // eprintln!("{:#?}", query_handle.selected_contents());
+        let mut num_rows = 0;
         for batch in query_handle.into_batch_iter() {
-            eprintln!("{batch}");
+            num_rows += batch.num_rows();
         }
+        eprintln!("read {num_rows} in {:?}", now.elapsed());
     }
 
     Ok(())
```

```
time cargo r -p re_dataframe --release --example query -- /tmp/tmp_40000.rrd
time cargo r -p re_dataframe --release --example query -- /tmp/tmp_80000.rrd

...etc
```

closed 2024-10-17 by emilk