rerun-io/rerun #7822: VideoFrameReference should have an extension constructor for `seconds=`

jleibs, 2024-10-17:
The default signature doesn't make it clear:
```
class VideoFrameReference(
    timestamp: VideoTimestampLike,
    *,
    video_reference: EntityPathLike | None = None
)

Create a new instance of the VideoFrameReference archetype.

Parameters
timestamp
References the closest video frame to this timestamp.

Note that this uses the closest video frame instead of the latest at this timestamp in order to be more forgiving of rounding errors for inprecise timestamp types.
```

I would recommend we disallow casting from `int` on `VideoTimestampLike` and add optional kwargs such as:
```
VideoFrameReference(seconds=12.4)
VideoFrameReference(nanos=124000000)
```

Wumpf, 2024-10-18:
the named arguments make sense to me but not allowing downcasting is weird for the batch constructor imho: `send_columns` should absolutely be able to take just an np array of integers

(also disallowing that is a breaking change, making this no longer eligible for a patch release)

jleibs, 2024-10-18:
I think the issue is that for people working with video seconds is way more common than nanoseconds.

Accidentally just doing `VideoFrameReferance(timestamp =1.0)` is an annoying footgun.

At a minimum that should produce an error for floats.

commit c23e81e1 (2024-10-22): Allow passing seconds/nanoseconds to `VideoFrameReference` archetype (#7833)

### What

* Fixes #7822

Also, passing a float directly will trigger a warning now.

### Checklist
* [x] I have read and agree to [Contributor
Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and
the [Code of
Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested the web demo (if applicable):
* Using examples from latest `main` build:
[rerun.io/viewer](https://rerun.io/viewer/pr/7833?manifest_url=https://a [... cut]
closed 2024-10-22 by Wumpf