rerun-io/rerun #7877: MVP for the Map view

abey79, 2024-10-23:
**Absolute "must" for the first release**:
- [x] Visible time range
- [ ] https://github.com/rerun-io/rerun/issues/7849
- [ ] https://github.com/rerun-io/rerun/issues/7872
- [ ] `GeoLineStrip` archetype
- [ ] Interactive entities: selectable and hover tooltip

**OK not to have for the first release**:
- [ ] `GeoPolygon` archetype
- [ ] `Altitude` component on all `GeoX` archetypes
- [ ] Leaner dependency tree for `walkers`

gavrelina, 2024-10-30:
thoughts and ideas! 
- https://github.com/podusowski/walkers/issues/210 add support to pan with the trackpad
- allow for transparency of the point styles (within a map context the layer of map actually has important information, so it's great to allow users flexibility to peek it once they need it)
- think of a seamless and easy flow to add mapbox API keys 
- I'd propose to change the default style of dots to have some kind of tiny shadow under, because maps are usually very busy with lots of colors and signs already, and now the dots are easily blending in with the rest. so would be cool to think how can we can make it standing out better by default. not urgent, but a nice improvement with time. 
- I can also imagine that being able to define different symbols and not only dots would come in handy.
- and a legend! 
- and adding labels! 

(okay I will stop here, I think the #1 is the prio, then the API key flow, and the rest can be added as we go)

Wumpf, 2024-11-15:
everything in the must-have list got done!

closed 2024-11-15 by Wumpf