rerun-io/rerun #7883: C++ SDK should enforce client to use cxx 17 standard

pmoulon, 2024-10-24:
**Describe the bug**

Using the C++ SDK in the c++ project

Compiler saying:
```
 error: ‘byte’ is not a member of ‘std’
```

See here byte required c++ 17 to be activated, not sure the rerun-sdk is enforcing the client to have this cxx standard enforced
https://en.cppreference.com/w/cpp/types/byte

**Rerun version**
0.19

**Additional context**
Issue first discovered here https://github.com/openMVG/openMVG/issues/2368

Wumpf, 2024-10-24:
Status quo is that we're setting the property for the rerun target itself here
https://github.com/rerun-io/rerun/blob/main/rerun_cpp/CMakeLists.txt#L30
and document that this is something that needs to be done for anyone using Rerun
https://rerun.io/docs/getting-started/quick-start/cpp#using-rerun-with-cmake

I think we need to instead use `target_compile_features` with `PUBLIC` to make it possible to "infect" any target using Rerun.
We need to test though if a user then can still use C++20! If that overrides it, we need to keep things as they are.

https://cmake.org/cmake/help/latest/manual/cmake-compile-features.7.html#requiring-language-standards says
> For example, if C++ 11 features are used extensively in a project's header files, then clients must use a compiler mode that is no less than C++ 11.

so this should be doing what we want

pmoulon, 2024-10-24:
Thank you @Wumpf for the feedback and details!

closed 2025-02-04 by teh-cmc