rerun-io/rerun #7916: Make the nodes draggable in the view

abey79, 2024-10-28:
Manually dragging the nodes in the graph view is useful both for fine-tuning the presentation and to interact with the (future) auto-layout algorithm for analysis purposes.

A correct implementation should transcribe the drag-and-drop interaction in the view into `Position2D` component overrides. However, for this to happen, we need per-instance override semantics.

**Possible solutions**:
- The override component contains `null` for all instances but those that have been manually dragged by the user. I _believe_ that this would not currently work, as the `null` value would also override which ever value was logged in store. If so, we could globally change the override semantics such that `null` means "do not override this instance".
- A variant of that would be to encode the null-instance override semantics (does `null` override or not) in a component tag.
- The per-instance override could be implemented at view level by having an `ManuallySet: [bool]` component, to track which instances of the `Position2D` instances should be considered. Insanely hacky, but doable without changing our data model.
- A slightly cleaner, more extensible variant would be to use tagged component to maintain multiple overrides for `Position2D`. One would be tagged with `manually_set`. In the future, another one could be `dynamic_auto_layout`, to store the intermediate states of the force-based auto-layout, should we want to expose them to the user across several frames.
- other?

Another aspect of this is how poor the current override UI is with non-mono-component (although in this case this is not a limiting factor strictly speaking, as the instance edit would happen via in-view UI interaction).

teh-cmc, 2024-10-28:
> A correct implementation should transcribe the drag-and-drop interaction in the view into `Position2D` component overrides. However, for this to happen, we need per-instance override semantics.

(I'm not familiar with the data model behind the new graph stuff)

Why do we need per-instance overrides at all? I.e. why is overriding all the positions in the position batch everytime any node is moved around not an option?

abey79, 2024-10-28:
> (I'm not familiar with the data model behind the new graph stuff)

Consider the `GraphNode` archetype has an optional `Position2D` component. Fallback to some kind of auto-layout (which could initially be as stupid as "arrange all nodes on a circle"). We can ignore edges here.

> Why do we need per-instance overrides at all?

I think this is required for the cases where the user logged time-varying `Position2D`. When one node is moved around in the UI, it's now-overridden position should apply regardless of the position of the time cursor, and should only affect that particular node. For that to happen, the un-overridden instance should be left "empty" in the override, because there is no "previous value" to duplicate (the previous value being time-varying).

(With that said, I now realise that there is an alternative approach that's hacky w.r.t to rerun data model, but closer to the graph view semantics we already decided. I'm adding it to the list.)