rerun-io/rerun #7920: Per-instance component override

abey79, 2024-10-28:
It would some time be desirable to have per-instance component override. Examples:
- I want _this_ point (of a bigger point cloud) to be fuchsia.
- I want _this_ node (of a bigger graph) to be moved at that location (see https://github.com/rerun-io/rerun/issues/7916)

Doing so necessitates the ability to override values for a single component, which is not currently possible.

Some possible avenues are listed here.

### A) Duplicate existing state in the other instances

The idea here is to take whatever values the component has currently, mutate the value for the instance to be overridden, and store that as override.

This breaks any time the component value is time varying, so it can be used ONLY when:
- the component is not logged at all in the data store, or logged as static
- the fallback value is not time-varying (which is in general not true)

So probably not an acceptable approach.

### B) Use `null` in the override for instances which aren't overridden

The idea here is to have a component full of `null`, except for the instance(s) to override, and to alter the override semantics such that `null` mean "keep using whatever other value you have for this instance".

**Pros**: very close to the current model

**Cons**:
- only applicable to nullable components, which might force us to use nullability in places we'd rather avoid to
- cannot override an instance such that it's value is now `null`

### B2) Tag the `null` semantic along the component

Variant where a component tag informs about the meaning of `null` (either "override this instance with `null`" or "keep whatever value you have for this instance").

### C) Allow an arrow dictionary representation for override components

A more natural way to specify instance override is, for each such instance, provide the instance index and the corresponding instance value. This essentially is a dictionary, so we could use that representation for component overrides.

**Pros**:
- possibly the most "correct" representation
- a better fit to the UI we'd present

**Cons**:
- there is no longer a single canonical arrow representation for a given component

grtlr, 2024-10-28:
I have some questions regarding option _C_:

* Would this functionality be exposed in the logging SDK as well, or will this remain a UI-level feature? There are some great use cases for this in the logging API (for example updating values that represented by the labels of a graph). We would have to spend some time making this ergonomic for the user, which leads mie directly to:
* What happens if the number of nodes in the graph changes? If I understand correctly, the instance ID is not stable under insertion / deletions, so we would have to additional specify at which timepoint the change was valid.

abey79, 2024-10-28:
> * Would this functionality be exposed in the logging SDK as well, or will this remain a UI-level feature? There are some great use cases for this in the logging API (for example updating values that represented by the labels of a graph).

I would assume this is the logical place where (C) leads. This would be a geopolitical-scale change to our data model, though, so this would take quite some design work. Notably, arrow dict array first appeared during discussions around dataframe export. So yeah it can be a very wide question.
 
> * What happens if the number of nodes in the graph changes? If I understand correctly, the instance ID is not stable under insertion / deletions, so we would have to additional specify at which timepoint the change was valid.

This is why my current line of thought is that this approach is not the best for the specific case of the graph view and we should favour some kind of `ViewProperty` (see #7916).

The usefulness of having stable ID for "things" that might be stored under varying instance indices over time may certainly have usefulness beyond the graph view. The idea of supporting such a thing at the core data model level is interesting, but, again, would be a major change.

Related: the `KeypointId` component used to attribute ids to instances, such that these instance may be referred to (e.g. in `AnnotationContext`.

> * We probably have to revise the current clamping logic for this too?

If we are to rely more on instance index to refer to things, yes, probably.

grtlr, 2024-10-28:
That makes sense to me, thank you for the clarification!

I also think the view property approach described in #7916 is a good choice for the graph view for now, as this part of the data model is still marked as experimental 👍.