rerun-io/rerun #7923: Add "clone" for view containers to context menu

peter-mitrano-bg, 2024-10-28:
I often "clone" views, and cloning a view container (tabs, horizontal, etc) would be nice as well. For example, I have some mildly complex nested hierarchy for all entities logged with "/tsne" prefix and I want to clone it and change it to view for all entities logged with "/pca" prefix.

**Describe the solution you'd like**
a "clone" option in the context menu for view containers.

**Describe alternatives you've considered**
Right now I manually recreate the hierarchy of views/containers.

**Additional context**

Wumpf, 2026-10-26:
sure, what do you want to discuss? Should be fairly straight forward I figure? :)

donnyboy72, 2026-10-28:
Yeah it wasn't super hard, I added a Clone option for containers that also copies the views and containers inside them. The main reason I was reaching out is because I created a new file called duplicate.rs and changed the name of copy_view.rs to copy_contents.rs I didn't know if that was a big enough code change to need to discuss it with you guys. (Trying to follow the contributing guidelines this is my first time contributing)

Wumpf, 2026-10-28:
thanks, very considerate! sounds alright; curious whether you ended up unifying the add view/container commands to easily bulk add things or whether you had to add something new. Do put up a pr for this if you're ready:)

donnyboy72, 2026-10-29:
I added a DuplicateContainer command that is similar to duplicate_view but duplicating a container copies everything inside the container.