rerun-io/rerun #7939: Is there a better way to debug external data loader

NerdToMars, 2024-10-30:
here are the errors when I running a rerun like `rerun --serve --web-viewer --web-viewer-port 9999 /host/data/rosbag_data` :

```
Failed to decode external loader's output filepath=xxxx err=Failed to decode the options: Reserved bytes not zero
```

https://github.com/rerun-io/rerun/blob/97e944b91ca3e58956dde06d77f8269aa3e43089/crates/store/re_data_loader/src/loader_external.rs#L212

It must because of debug info! that I used inside the data loader, but it should not failed the entire loading process IMO.

Just feel a bit difficult to debug this kind of issue or is it possible to output stdout of the external data loader?

teh-cmc, 2024-10-30:
> It must because of debug info! that I used inside the data loader, but it should not failed the entire loading process IMO.

A dataloader can log anything it wishes on stderr, it won't interfere with the actual data (which is passed through stdout).

What's logged on stderr will be included in the error message shown by the viewer if the dataloader exits with an error status, see:
https://github.com/rerun-io/rerun/blob/97e944b91ca3e58956dde06d77f8269aa3e43089/crates/store/re_data_loader/src/loader_external.rs#L266

> Just feel a bit difficult to debug this kind of issue or is it possible to output stdout of the external data loader?

A dataloader is just a binary that spits data on stdout -- you can manually execute it (even through a debugger) and check its output yourself, as in:
```sh
./mydataloader | rerun -
./mydataloader | hexdump -C
```

teh-cmc, 2024-10-30:
(I think that answers the question, feel free to re-open otherwise)

closed 2024-10-30 by teh-cmc

NerdToMars, 2024-11-01:
ya, thanks a lot, it's turn out there is println!() in my function, after I remove the prinln!(), no error anymore.