rerun-io/rerun #7951: Rerun Web Visualizer Crash

nisseknudsen, 2024-10-30:
Hey there, I am using the `EncodedImage` object to log a bunch of jpeg images (at 30 fps). Occasionally my rerun web visualizer just crashes with this error:

```
Rerun has crashed.

panicked at /Users/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/zune-jpeg-0.4.11/src/mcu.rs:209:9:
assertion `left == right` failed
  left: 6071040
 right: 6220800
 ```
 
 
 It's always about the `left == right` assertion in that library. I have no idea where this is coming from.

I do not have different code paths to generate or log those jpeg images, so I'm wondering 🤔

nisseknudsen, 2024-10-30:
I changed my code to decode the `jpeg` into an ndarray and logging it as a regular `Image` archetype, but obviously this racks up memory consumption _very_ quickly, and the web visualizer starts discarding data at around 2GB.

(again, we're logging with 30 fps so not sure if this is a supported use case)

nisseknudsen, 2024-10-30:
I found something interesting:

when logging the `EncodedImage` from `jpeg`, when it crashes after around 10 seconds of solid logging, I see in the server logs now:

```
Corrupt JPEG data: premature end of data segment
```

When I take the `jpeg`, and transcode it to `png` and log it as `EncodedImage`, I do _not_ get a crash after those ca. 10 seconds, but I do get a bunch of visually corrupted images and stuttering in my logging timeline. I suspect it relates to my image source that produces corrupt jpeg images over time, and the rerun crash is just a symptom for it.

PNG EncodedImage:

https://github.com/user-attachments/assets/99452b60-eca4-4a31-a41c-08079eaab31a

When checking the logs, I do see now oddly this error:
```
Corrupt JPEG data: premature end of data segment
```

I can't say if this log is produced by `rerun` or by my `cv2.imdecode` method, though.

emilk, 2024-10-31:
Thanks for the report! If you can find a JPEG that consistently crashes Rerun, I would love to have it

emilk, 2024-10-31:
The offending assert was removed from `zune-jpeg` 10 months ago: https://github.com/etemesi254/zune-image/commit/1f943d81b1eb189535abfd2afac8f017c57164f3

emilk, 2024-10-31:
Fix coming in:
* https://github.com/rerun-io/rerun/pull/7952

closed 2024-10-31 by emilk

nisseknudsen, 2024-10-31:
Thanks ƒor the quick fix @emilk ! I unfortunately do not have the jpeg (it was a stream so didn't write it to disk). If I run into this again, I'll see to write the image out for you.