rerun-io/rerun #7969: Text issues on RK3588, RK3588S and RK3588S2 chipsets

DmitriYam, 2024-11-01:
Hi,

I would like to raise a bug report. Running rerun >= 0.17.0 on RK3588 SoCs seem to cause issues. Texts do not show up when running rerun. Attached below are some more information as well as a video demonstrating the issue. The video captured on a radxa rock 5c running rerun 0.17.0. Tested on both Wayland and X11. 

Boards tested

Radxa ROCK 5A - RK3588S
Radxa ROCK 5C - RK3588S2
Nano PC T6 - RK3588

**To Reproduce**
Steps to reproduce the behavior:
1. Install ubuntu on the above boards from the at the following link: https://joshua-riek.github.io/ubuntu-rockchip-download/
2. Install rerun 0.17.0 or above. 
3. run rerun

**Expected behavior**
Rerun examples are expected to normally.

**Desktop (please complete the following information):**
Linux pi-calibration-2 5.10.0-1012-rockchip #12-Ubuntu SMP Wed Aug 14 22:22:22 UTC 2024 aarch64 aarch64 aarch64 GNU/Linux
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.4 LTS
Release:	22.04
Codename:	jammy

**Rerun version**
rerun-cli 0.17.0 [rustc 1.76.0 (07dca489a 2024-02-04), LLVM 17.0.6] aarch64-unknown-linux-gnu release-0.17.0 d65ca50, built 2024-07-08T13:51:29Z

Screenshot of system info:
[image]

Video demonstrating issue:
[Screencast from 01-11-2024 18:34:20.webm](https://github.com/user-attachments/assets/38112f7b-77bc-4da5-b157-6fcd85498720)

Wumpf, 2024-11-01:
Rendering issues looks suspiciously similar to the ones in
* https://github.com/rerun-io/rerun/issues/7177

Given that you explicitely set the GL renderer, I reckon the system runs into issues when not instructing it so explicitly?
Is there a Vulkan driver as well?
We don't yet do patches for any version older than the latest release, could you also check on 0.19 as well?
(there's a slim chances recent updates to wgpu, our rendering backend library, may change the situation)

DmitriYam, 2024-11-04:
Yes, the system runs into similar issues when not instructing explicitly. 
There seems to be no vulkan driver support. https://github.com/Joshua-Riek/ubuntu-rockchip/issues/617 
The issues persists on 0.19 as well.

I tested rerun 0.15.1 on the same boards using explicitly setting the GL renderer and it works fine.

Elabajaba, 2026-03-24:
RK3588 does have a vulkan 1.4 conformant driver now (PanVK, https://www.khronos.org/conformance/adopters/conformant-products#submission_939), though it is lacking quite a few extensions (eg. dxvk doesn't work because it needs unsupported extensions).

If someone wants they can compare https://mesamatrix.net/ vs the ones required by wgpu/webgpu/rerun.