rerun-io/rerun #7971: Provide an LLM export of our docs

jprochazk, 2024-11-01:
There's a recent trend for libraries to provide a file that contains enough context about the library for an LLM to be able to act as an assistant for using that library. The idea behind it being a file is that you can provide it to any LLM that supports such an import, and it also makes distribution very easy.

In our case we'd probably build it in the "push to docs-latest" job, and have a button somewhere on our website to download it.

There are some tools that can generate such an export, which we should try out. One such example is [repopack](https://github.com/yamadashy/repopack).

If none of the available tools are suitable, we'd have to develop something on top of the existing search index build infrastructure. The LLM would likely want a lot more information about the available symbols, their fields/parameters/return types, all the doc comments, full import paths per-language, etc., plus all our examples/snippets. We could limit it to Python-only in the MVP.

pablovela5620, 2024-12-04:
I think this would also be useful
https://llmstxt.org/
being able to use chatgpt/claude for help with rerun would be super helpful

jprochazk, 2024-12-04:
> I think this would also be useful
> [llmstxt.org](https://llmstxt.org/)

We have a WIP of `llms.txt`, but I haven't been able to get it to work with Claude due to size limits. ChatGPT works somewhat, here's a GPT I made using the WIP export: https://chatgpt.com/g/g-674702fde5948191a810bdf73370b6eb-rerun-gpt - it really likes to hallucinate outdated usage examples due to its training data being pretty far behind the current version

jprochazk, 2024-12-04:
I mean that the code it generates uses APIs that no longer exist, such as `rr.log_image` instead of `rr.log("...", rr.Image(...))`. The generated `llms.txt` file is pretty much just the markdown version of `rerun.io/docs`, though.