rerun-io/rerun #7973: Video opened through rerun looks distorted and has wrong colors

pablovela5620, 2024-11-01:
**Describe the bug**

Something went wrong with showing an av1 encoded video. Here are is what it looks like in rerun and what it looks like in vlc

## Rerun
https://github.com/user-attachments/assets/b0a692c9-2205-46b5-9964-29065db32faa

## VLC
https://github.com/user-attachments/assets/559d01f1-9d9c-42fa-8ad8-46ee273a24d3

**To Reproduce**
Steps to reproduce the behavior:
1. Open video via `rerun $Path to video`

here is the video that I used that caused problems
https://huggingface.co/datasets/pablovela5620/rerun-bug-video/blob/main/combined-av1.mp4

**Desktop (please complete the following information):**
 - OS: Mac OS 15.1 (24B83)

**Rerun version**

rerun-cli 0.19.0 [rustc 1.79.0 (129f3b996 2024-06-10), LLVM 18.1.7] aarch64-apple-darwin, built 2024-10-21T16:15:46Z
Video features: av1 nasm

**Additional context**
This was after I merged some mp4 files and reenocded them to av1 using the following command
```
ffmpeg -i combined.mp4 -c:v libsvtav1 -preset 10 -crf 35 -c:a copy combined-av1.mp4
```

emilk, 2024-11-02:
I get the same problem on the native build:
[image]

The web-viewer plays it perfectly though

emilk, 2024-11-02:
Oh shit, I missed this:

[image]

This is therefore a duplicate of:
* #7594

It makes me think we should just refuse to play such videos though, instead of showing something so obviously wrong

emilk, 2024-11-02:
I'll add a warning to the selection panel, and maybe see if there is an easy way to at least get _almost_ correct colors

pablovela5620, 2024-11-02:
For some reason I also didn't see the pop up on my mac, when I tested on my linux machine it showed up!

closed 2024-11-02 by emilk