rerun-io/rerun #7990: Python API to administer (remove) applications, recordings and entities

Famok, 2024-11-04:
**Is your feature request related to a problem? Please describe.**
Currently, it is not possible to administer applications, recordings, or entities via the Python API. When working with multiple recordings or wanting to update recordings, I would like the ability to remove them programmatically. At present, the only way to remove them is by manually clicking in the viewer. 
Additionally, when working with numerous recordings, the viewer may end up with partially deleted recordings, as it automatically drops the oldest datasets.

**Describe the solution you'd like**
A few api commands to administer recordings.

Commands to get the current state:
```
rr.get_applications() -> returns all application ids
rr.get_recordings(<application_id>) -> returns all recording ids for the given application id
rr.get_entities(<application_id>,<recording_id>) -> returns all entity ids for given application and recording id
```

Commands to delete specific things:
```
rr.delete_application(<application_id>)
rr.delete_recording(<application_id>,<recording_id>)
rr.delete_entity_path(<application_id>,<recording_id>,<entity_path>)
```

Commands to delete more / all things:
```
rr.delete_all_applications()
rr.delete_applications([list of ids])
...

```

**Describe alternatives you've considered**
There seems to be a pull request here: [](https://github.com/rerun-io/rerun/pull/6588) 
and an issue here [](https://github.com/rerun-io/rerun/issues/1329) but I can't find any documentation on drop_entity_path in the python api.

All these things exist in the viewer, so I hope this could be easy to implement.

Wumpf, 2024-11-06:
Related to
* https://github.com/rerun-io/rerun/issues/1329

Wumpf, 2025-11-11:
It's not called out on any short term roadmap, but that said I think as we move to a client/server everything model we're getting a lot closer to a nice way of expressing this: as we're starting to build out our commercial dataplatform we _also_ already (more or less stealthy) published an open source server that speaks the same protocol. The idea is that in the hopefully near future the Rerun viewer _always_ connects to a server and then we can have operations on that server like removing recordings etc.
Obviously, connecting to a local Rerun Viewer today isn't all that different from connecting to some server/store of recordings, but this has a lot to do with how we want to model additional data/viewer communication in the future, which is why I bring this up.

cc: @abey79 having a "remove recording" endpoint on OSS server should be pretty easy if we don't have it already, right? 🤔

Famok, 2025-11-12:
Is there any documentation on the server I could read? Seems like something i would like to use!

Famok, 2025-11-16:
Thanks for the link. As far as I can understand, the OSS server is there to provide files for the viewer?
Or is there some more information on what it is for?

abey79, 2025-11-17:
> Thanks for the link. As far as I can understand, the OSS server is there to provide files for the viewer? Or is there some more information on what it is for?

This is not yet fully reflected by our landing page (update is in progress). The data model of the Rerun server (which we call "redap" for rerun data protocol), as well as the querying capability we're building in the SDK, are geared toward implementation robotics/physical AI data pipelines. The OSS server is a "basic" implementation of redap, with limitations in scaling and persistence (e.g. all data is held in RAM), but is otherwise fully functional and can serve with, e.g., local-scale pipelines, prototyping and CI/test automation. Our upcoming managed service will provide a scalable service over the same protocol.