rerun-io/rerun #8018: Nightly failing: "terminate called without an active exception"

teh-cmc, 2024-11-06:
:unamused:

abey79, 2024-11-06:
Full CI run on the offending PR: https://github.com/rerun-io/rerun/actions/runs/11704123409

teh-cmc, 2024-11-06:
Still fails just as much on Rust 1.80: https://github.com/rerun-io/rerun/actions/runs/11705944608/job/32602393114?pr=8019

teh-cmc, 2024-11-06:
Repro'd locally:
```
#0  0x00007a019fe533f4 in ?? () from /usr/lib/libc.so.6
#1  0x00007a019fdfa120 in raise () from /usr/lib/libc.so.6
#2  0x00007a019fde14c3 in abort () from /usr/lib/libc.so.6
#3  0x00007a016fcd5fae in __gnu_cxx::__verbose_terminate_handler () at ../../../../libstdc++-v3/libsupc++/vterminate.cc:95
#4  0x00007a016fcd44f2 in __cxxabiv1::__terminate (handler=<optimized out>) at ../../../../libstdc++-v3/libsupc++/eh_terminate.cc:48
#5  0x00007a016fcce2f3 in std::terminate () at ../../../../libstdc++-v3/libsupc++/eh_terminate.cc:58
#6  0x00007a016fcd41c1 in __cxxabiv1::__gxx_personality_v0 (version=<optimized out>, actions=10, exception_class=0, ue_header=0x79ffb8600d30, context=0x79ffb85feff0)
    at ../../../../libstdc++-v3/libsupc++/eh_personality.cc:685
#7  0x00007a019ea03f35 in _Unwind_ForcedUnwind_Phase2 (exc=exc@entry=0x79ffb8600d30, context=context@entry=0x79ffb85feff0, frames_p=frames_p@entry=0x79ffb85feef8) at ../../../libgcc/unwind.inc:183
#8  0x00007a019ea0429c in _Unwind_ForcedUnwind (exc=0x79ffb8600d30, stop=0x7a019fe59e80, stop_argument=<optimized out>) at ../../../libgcc/unwind.inc:218
#9  0x00007a019fe5a045 in ?? () from /usr/lib/libc.so.6
#10 0x00007a019fe52732 in pthread_exit () from /usr/lib/libc.so.6
#11 0x00005687e67260b7 in PyThread_exit_thread () at /usr/local/src/conda/python-3.11.10/Python/thread_pthread.h:366
#12 0x00005687e668e990 in take_gil (tstate=tstate@entry=0x79ff3c0012f0) at /usr/local/src/conda/python-3.11.10/Python/ceval_gil.h:207
#13 0x00005687e677bac2 in PyEval_RestoreThread (tstate=tstate@entry=0x79ff3c0012f0) at /usr/local/src/conda/python-3.11.10/Python/ceval.c:521
#14 0x00005687e685f91d in PyGILState_Ensure () at /usr/local/src/conda/python-3.11.10/Python/pystate.c:1712
#15 0x00007a017870cc9e in arrow::py::NumPyBuffer::~NumPyBuffer() () from /home/cmc/dev/rerun-io/rerun/.pixi/envs/py/lib/python3.11/site-packages/pyarrow/libarrow_python.so
#16 0x00007a01786e3073 in std::_Sp_counted_ptr_inplace<arrow::ArrayData, std::allocator<void>, (__gnu_cxx::_Lock_policy)2>::_M_dispose() ()
   from /home/cmc/dev/rerun-io/rerun/.pixi/envs/py/lib/python3.11/site-packages/pyarrow/libarrow_python.so
#17 0x00007a01786e304b in std::_Sp_counted_ptr_inplace<arrow::ArrayData, std::allocator<void>, (__gnu_cxx::_Lock_policy)2>::_M_dispose() ()
   from /home/cmc/dev/rerun-io/rerun/.pixi/envs/py/lib/python3.11/site-packages/pyarrow/libarrow_python.so
#18 0x00007a008e768bf7 in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release() () from /home/cmc/dev/rerun-io/rerun/.pixi/envs/py/lib/python3.11/site-packages/pyarrow/../../../libarrow.so.1800
#19 0x00007a008e960b3b in arrow::(anonymous namespace)::ReleaseExportedArray(ArrowArray*) () from /home/cmc/dev/rerun-io/rerun/.pixi/envs/py/lib/python3.11/site-packages/pyarrow/../../../libarrow.so.1800
#20 0x00007a0089c5db70 in core::ptr::drop_in_place<dyn arrow_buffer::alloc::Allocation> () at /rustc/129f3b9964af4d4a709d1383930ade12dfe7c081/library/core/src/ptr/mod.rs:514
#21 alloc::sync::Arc<dyn arrow_buffer::alloc::Allocation, alloc::alloc::Global>::drop_slow<dyn arrow_buffer::alloc::Allocation, alloc::alloc::Global> (self=<optimized out>)
    at /rustc/129f3b9964af4d4a709d1383930ade12dfe7c081/library/alloc/src/sync.rs:1812
#22 0x00007a0089c5dfac in alloc::sync::{impl#34}::drop<dyn arrow_buffer::alloc::Allocation, alloc::alloc::Global> (self=0x34b9f0) at /rustc/129f3b9964af4d4a709d1383930ade12dfe7c081/library/alloc/src/sync.rs:2470
#23 core::ptr::drop_in_place<alloc::sync::Arc<dyn arrow_buffer::alloc::Allocation, alloc::alloc::Global>> () at /rustc/129f3b9964af4d4a709d1383930ade12dfe7c081/library/core/src/ptr/mod.rs:514
#24 core::ptr::drop_in_place<arrow_buffer::alloc::Deallocation> () at /rustc/129f3b9964af4d4a709d1383930ade12dfe7c081/library/core/src/ptr/mod.rs:514
#25 core::ptr::drop_in_place<arrow_buffer::bytes::Bytes> () at /rustc/129f3b9964af4d4a709d1383930ade12dfe7c081/library/core/src/ptr/mod.rs:514
#26 alloc::sync::Arc<arrow_buffer::bytes::Bytes, alloc::alloc::Global>::drop_slow<arrow_buffer::bytes::Bytes, alloc::alloc::Global> (self=<optimized out>)
    at /rustc/129f3b9964af4d4a709d1383930ade12dfe7c081/library/alloc/src/sync.rs:1812
#27 0x00007a0089c5dd54 in core::ptr::drop_in_place<foreign_vec::ForeignVec<re_arrow2::buffer::BytesAllocator, u32>> () at /rustc/129f3b9964af4d4a709d1383930ade12dfe7c081/library/core/src/ptr/mod.rs:514
#28 core::ptr::drop_in_place<re_arrow2::buffer::Bytes<u32>> () at /rustc/129f3b9964af4d4a709d1383930ade12dfe7c081/library/core/src/ptr/mod.rs:514
#29 alloc::sync::Arc<re_arrow2::buffer::Bytes<u32>, alloc::alloc::Global>::drop_slow<re_arrow2::buffer::Bytes<u32>, alloc::alloc::Global> (self=<optimized out>)
    at /rustc/129f3b9964af4d4a709d1383930ade12dfe7c081/library/alloc/src/sync.rs:1812
#30 0x00007a0089c6e55f in alloc::sync::{impl#34}::drop<re_arrow2::buffer::Bytes<f32>, alloc::alloc::Global> (self=0x34b9f0) at /rustc/129f3b9964af4d4a709d1383930ade12dfe7c081/library/alloc/src/sync.rs:2470
#31 core::ptr::drop_in_place<alloc::sync::Arc<re_arrow2::buffer::Bytes<f32>, alloc::alloc::Global>> () at /rustc/129f3b9964af4d4a709d1383930ade12dfe7c081/library/core/src/ptr/mod.rs:514
#32 core::ptr::drop_in_place<re_arrow2::buffer::immutable::Buffer<f32>> () at /rustc/129f3b9964af4d4a709d1383930ade12dfe7c081/library/core/src/ptr/mod.rs:514
[... cut ...]
#58 0x00007a0089e849cb in alloc::boxed::{impl#48}::call_once<(), dyn core::ops::function::FnOnce<(), Output=()>, alloc::alloc::Global> () at library/alloc/src/boxed.rs:2022
#59 alloc::boxed::{impl#48}::call_once<(), alloc::boxed::Box<dyn core::ops::function::FnOnce<(), Output=()>, alloc::alloc::Global>, alloc::alloc::Global> () at library/alloc/src/boxed.rs:2022
#60 std::sys::pal::unix::thread::{impl#2}::new::thread_start () at library/std/src/sys/pal/unix/thread.rs:108
#61 0x00007a019fe5139d in ?? () from /usr/lib/libc.so.6
#62 0x00007a019fed649c in ?? () from /usr/lib/libc.so.6
```

teh-cmc, 2024-11-06:
Ye that was me in https://github.com/rerun-io/rerun/commit/61c23afbe911f46629cd084bf5a129385e47321e :smile: 

Just need to figure out how something has managed to escape the garbage queue, it should be impossible...

teh-cmc, 2024-11-06:
Ok I'm 99% sure it's because the dataframe APIs are now able to load recordings/chunks without going through the log functions and therefore bypass all these protections.

teh-cmc, 2024-11-07:
The issue is definitely timing based and, as far as I can tell, has always been there. This is not the result of some recently added code.

The reason it now fails consistently on CI for @abey79 's branch is because that branch introduced a bunch of new unit tests, which changed the timings of the test runner in a way that always triggers the issue.
Removing these tests makes the CI pass again on that branch.

I used to be able to reproduce the failing timing locally on my machine (that's how I grabbed the coredump posted above), but ever since I rebooted my machine this morning, I haven't been able to hit it again :zany_face:

teh-cmc, 2024-11-07:
There we go.

Take the following code:
```python
import rerun as rr

iterations = range(0, 1)

for i in iterations:
    print(f"{i}")
    rr.init("rerun_example_points3d")
    rr.log("points", rr.Points3D([[0, 0, 0], [1, 1, 1]]))
```

* `iterations = range(0, 1)` works.
* `iterations = range(0, 2)` up to range(0, 7) fails with `terminate called without an active exception`.
* `iterations = range(0, 8)` and anything above will deadlock.

abey79, 2024-11-07:
Funnily this would be prevented if test were isolated as per #8023

commit b68176dd (2024-11-07): Fix python SDK's shutdown unsafely dropping cross-FFI resources (#8038)

* Fixes #8018
closed 2024-11-07 by teh-cmc