rerun-io/rerun #8025: Make `rec.log_file`  able to overwrite `entity_path` to make it easier to handle them without being dependant on the filename which can be ugly

haixuanTao, 2024-11-07:
Make `rec.log_file`  able to overwrite `entity_path` to make it easier to handle them without being dependant on the filename which can be ugly.

**Describe the annoyance**

Let say we want to create an app that let users log their urdf file. let's say the file name is going to be `piper.urdf`. 

Currently, the only thing we can do on the entity_path is prefix it with something like `production_piper.urdf` 

Now the thing is that this entity is refering to something i personally call `bob`, well I can't rename this entity_path bob. I can only rename the file. But maybe I want to use this file for another entity call `mary`. 

**Expected behavior**
Add a function to `log_file` but that can actually overwrite the `entity_path`.

Maybe I didn't find it.

teh-cmc, 2024-11-07:
All `rec.log_file` does is transfer control to a dataloader. That dataloader is then free to log whatever it wants, to whatever entity path it wants -- Rerun itself has no say in any of this.

I assume you're referring to our URDF example in this specific case. This examples uses the filename as its entity path because it explicitly chooses to:
https://github.com/rerun-io/rerun-loader-python-example-urdf/blob/539252acc50771829f7e2aacef27d6cb2cbd7928/rerun_loader_urdf.py#L313-L316
```py
if args.entity_path_prefix is not None:
    prefix = args.entity_path_prefix
else:
    prefix = os.path.basename(args.filepath)
```

You should fork/copy-paste this dataloader and modify its code in order to make it behave however you'd like it to, you'll have full control from there.
Dataloaders are just cheap, arbitrary, external scripts -- they are designed that way so that they can be easily fork to accommodate to anyone's needs.

See also: https://rerun.io/docs/reference/data-loaders/overview

---

On a related note: even if we were to add an `--entity-path` flag, you would still need to fork and modify the dataloader itself in order for it to actually support that flag.

What I would like us to have in the future is server-side rewrite rules, so that you could pass e.g. regex-based remappings in `log_file()`, and have the Rerun database monkey-patch the dataloaded data so that it follows these rules,e.g.:
```
# entity path `xxx` will by remapped to `yyy` by monkey patching the data directly on the server-side.
rr.log_file("model.urdf", [EntityPathRemapRule("s/xxx/yyy/g")])
```
But that's a much, much bigger can of worms -- for now everything is handled client-side, by the dataloader itself.

closed 2024-11-07 by teh-cmc

haixuanTao, 2024-11-07:
Got it. Thanks for the detailed explanation!

Will patch rerun urdf loader in that case.