rerun-io/rerun #8039: `log_file_from_path` not logging to the recording parameter

EtaLoop, 2024-11-07:
**Describe the bug**
Here, I want to log a supported file (obj) in the "my_app" recording. But with `log_file_from_path`, this logs to a "/path/to/file/file.obj" recording.
Removing `log_file_from_path` make it logs correctly to "my_app". So, my guess was that, it renames the recording to the file path.

Is it me doing something wrong here, or no?

I'm on Rerun 0.19.1.
**Code :**
```
import rerun as rr

def main():
    rec = rr.init("my_app", spawn=True)
    
    rr.log(
        "triangle",
        rr.Mesh3D(
            vertex_positions=[[0.0, 1.0, 0.0], [1.0, 0.0, 0.0], [0.0, 0.0, 0.0]],
            vertex_normals=[0.0, 0.0, 1.0],
            vertex_colors=[[0, 0, 255], [0, 255, 0], [255, 0, 0]],
            triangle_indices=[2, 1, 0],
        ),
        recording=rec,
    )
    
    rr.log_file_from_path(file_path="/pah/to/file/file.obj", recording=rec)
```

teh-cmc, 2024-11-07:
You are _not_ doing anything wrong, this is one of the many subtle data-loading bugs that were fixed in:
* https://github.com/rerun-io/rerun/pull/7966

These fixes will ship in 0.20, which is right around the corner (ish).

If you cannot wait for 0.20, you can try using one of our nightly builds instead: https://github.com/rerun-io/rerun/releases/tag/prerelease

closed 2024-11-07 by teh-cmc

EtaLoop, 2024-11-07:
Oh, great ! Thanks

qianyizhang, 2024-11-19:
> You are _not_ doing anything wrong, this is one of the many subtle data-loading bugs that were fixed in:
> 
> * [Consistent open/import/log_file behaviors in all common scenarios #7966](https://github.com/rerun-io/rerun/pull/7966)
> 
> These fixes will ship in 0.20, which is right around the corner (ish).
> 
> If you cannot wait for 0.20, you can try using one of our nightly builds instead: https://github.com/rerun-io/rerun/releases/tag/prerelease

I am working with 0.20, still doesn't overwrite the application_id as intended 😣

qianyizhang, 2024-11-19:
another related issue https://github.com/rerun-io/rerun/issues/8167

teh-cmc, 2024-11-19:
> I am working with 0.20, still doesn't overwrite the application_id as intended 😣

I can confirm the script posted in this issue works as expected on 0.20:
[image]